Qucs/qucs #155: Regression Qucsator

guitorri, 2015-01-03:
After recent merges the example [qucs-test/testsuite/DC_TR_SW_spice_BFR520_prj]() started failing.

See logs below:

0.0.18 (debug mode)
https://gist.github.com/guitorri/a87a47194311ba133705

0.0.19 (after forguillem_cleanup_20141226 merge)
https://gist.github.com/guitorri/8c784618f012a17ecc14

Observations:
- internal component names are mangled (before / now)

   ```
   L:BFR520_cir.BFR520_1.X1.L1 _net1 BFR520_cir.BFR520_1.X1._net4 L="3.5e-10"
   L:BFR520_cir.BFR520_1..X1.L1 _net1 BFR520_cir.9?{?.X1._net4 L="3.5e-10"
   ```

- the last DC solutions are converging faster now.
- the TR is failing to start

guitorri, 2015-01-03:
@bastien-roucaries can you take a took at it?

bastien-roucaries, 2015-01-04:
It is commit c9694eb6595d767c55d76cb9416c2b633b8a4814 (by bissection)

bastien-roucaries, 2015-01-04:
Sorry this is this one:
c49583263bdb4d98076c6a21fbf125c24dfe8516 is the first bad commit

commit c49583263bdb4d98076c6a21fbf125c24dfe8516
Author: Bastien ROUCARIÈS <roucaries.bastien@gmail.com>
Date:   Fri Dec 26 18:41:13 2014 +0100

    Use std::string for text concatenation in strlist

:040000 040000 8ca013259ecce883515587b5f143fd18ba10c5e1 6b2a4a2aa3360cdf3f7461a891a5a54fc93a27dc M      qucs-core

bastien-roucaries, 2015-01-04:
Reverting work but it means that lifetime of object is not really clean and moreover it means massive leak...

Bastien

guitorri, 2015-01-04:
Can you point to the leak? Did you run it with Valgrind?

bastien-roucaries, 2015-01-04:
I run under valgrind. The goal of this patch was to clean lifetime of string... And it fail because code does not honor const contract.

closed 2015-01-30 by guitorri