Qucs/qucs #161: Shall we remove `qucs-filter-v2`?

guitorri, 2015-01-07:
The `qucs-filter-v2` seems to have models that are not in `qucs-filter`. However in '-v2' some tabs are missing (#75)  and it seems to be unfinished. 

Furthermore it is not being packaged, it only builds with `-stdlib=libstc++` on OSX (default to `libc++`). 

See also here for issues on BSD: https://sourceforge.net/p/qucs/mailman/message/32825591/

Is someone interested in having a look at it and salvage what can be reused in `qucs-filter`?

ra3xdh, 2015-01-08:
I think qucs-filter-v2 should not be romoved right now. It contains some interesting filter transfer function approximations (Legendre and Blinchikoff). We should backport them to qucs-filter. After that we can remove qucs-filter-v2. Qucs-filter and qucs-filter-v2 should be merged.

yodalee, 2015-01-10:
Maybe we can first remove it from compile rule, but leave the data there so we can keep moving our building system forward. qucs-filter-v2 seems containing some dated function declare that give out warning every time I compile.

yodalee, 2015-04-13:
Remove filter-v2 from build system, wait extracting useful part.

guitorri, 2016-01-26:
Portions of filter-v2 are not portable. The code is no being build, it will bit rot further.
Let us see if we can merge/refactor the useful parts by 0.0.20.