Qucs/qucs #164: Autotools qucs-core parallel build race condition, error.

guitorri, 2015-01-07:
Some times the `qucs-core` built will error out during a `make -j 8` while running ADMS.
Something on the dependencies is wrong. The compiler wants the verilob-a auto-generated [.h, .cpp] files that are not yet available.

felix-salfelder, 2015-06-29:
`libverilog_la_SOURCES` lists a lot of .cpp files that are not there. it might help to add them to `BUILT_SOURCES` to workaround this, http://www.gnu.org/software/automake/manual/html_node/Built-Sources-Example.html.

actually, the .va files are the source, so *these* should be listed as the SOURCES, see http://www.gnu.org/software/automake/manual/html_node/Suffixes.html.

(that's all I know, did not run into the described race)

nvdl, 2015-06-29:
Error confirmed. I thought, something was wrong on my side.

guitorri, 2015-06-29:
The `BUILT_SOURCES` is a good hint.

The `.va` and the `.xml` files are both the sources. Both should be set as `SOURCES`. Right now I don't thing it will rebuild if these files are modified. The dependency check on these don not seem to be working.

Concerning the race condition, admsXml is called for each XML script to generate code. Running a parallel make, it _may_ create the `.core.[h,cpp]` _before_ the `.analogfunction.[h.cpp]`. When it tries to compile the objects for `.core.cpp` it fails to find the included `analogfunction.h` which is still missing...

One way to solve this is to pass all the XML scripts in one call to `admsXml`. This would generate all the headers and sources necessary for a given `.va` in one shot. I did that for [CMake](https://github.com/Qucs/qucs/blob/master/qucs-core/src/components/verilog/CMakeLists.txt) (no particular reason, just to have a shorter CMake). Note that there is an order on which the XML files are passed to `admsXml`. 

Can we write some sort of macro with Automake to simplify/shorten the `verilog/Makefile.am`?

felix-salfelder, 2015-06-29:
> The .va and the .xml files are both the sources

no. think of each item as a compilation unit. the xml files don't belong there, these are just prerequisites for the code generator. from the manal:

> The variable hello_SOURCES is used to specify which source files get built into an executable:
> hello_SOURCES = hello.c version.c getopt.c getopt1.c getopt.h system.h 
> This causes each mentioned .c file to be compiled into the corresponding .o. Then all are linked to  produce hello. 

> Can we write some sort of macro with Automake to simplify/shorten the verilog/Makefile.am?

i'd first sort out the functionality, then simplify. this hasn't highest prioroty for me, so I don't insist on doing it.

felix-salfelder, 2015-08-16:
there's a functional implementation in #294. this PR addresses the deeper problem with the admsXml invocation. right now, random files (unpredictable names) are created from a `file.va`, instead of just `file.cpp`. with the proposed changes, `file.o` can be created from `file.cpp` from `file.va`. this is easy for most build systems, it works with a simple Makefile.am (included).

(the existence of various extra files and the potential name clashes remain untouched)

commit 35114c4d (2015-09-28): Merge pull request #294 from felix-salfelder/admsrules

trying to simplify .va compilation rules and fix #164
closed 2015-09-28 by guitorri