Qucs/qucs #171: Build fails on Windows

in3otd, 2015-01-10:
Trying to build `Qucs` on Windows 7 following the [instruction on the Wiki] (https://github.com/Qucs/qucs/wiki/Build-Windows) (this is the first time I try), I get an error on `qucs-filter` : 

````
g++ -DHAVE_CONFIG_H -I. -I..   -I/c/Qt/4.8.6//include -I/c/Qt/4.8.6//include/Qt
-I/c/Qt/4.8.6//include/QtGui -I/c/Qt/4.8.6//include/QtCore -I/c/Qt/4.8.6//includ
e/QtSvg -I/c/Qt/4.8.6//include/QtXml -I/c/Qt/4.8.6//include/QtScript -I/c/Qt/4.8
.6//include/Qt3Support -DQT_DEBUG -DQT3_SUPPORT -DQT_THREAD_SUPPORT -D_REENTRANT
 -mthreads -DQT_DLL -DUNICODE   -g -O2 -pipe -fno-exceptions -W -Wall -std=c++0x
 -O0 -MT eqn_filter.o -MD -MP -MF .deps/eqn_filter.Tpo -c -o eqn_filter.o eqn_fi
lter.cpp
eqn_filter.cpp: In static member function 'static QString* Equation_Filter::crea
teSchematic(tFilter*)':
eqn_filter.cpp:71:16: error: 'M_PI' was not declared in this scope
   Omega *= 2.0*M_PI;   // angular frequency
                ^
make[2]: *** [eqn_filter.o] Error 1
make[2]: Leaving directory `/c/git/qucs/qucs/qucs-filter'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/c/git/qucs/qucs'
make: *** [all] Error 2
````
did someone else try to build on Windows recently?

in3otd, 2015-01-10:
oh, seems to be related to this http://ubuntuforums.org/showthread.php?t=583094 ; if I remove the `-std=c++0x` it compiles ok...  interesting "improvement" in C99 :astonished:

ra3xdh, 2015-01-10:
What is MinGW version? We moved project from `math.h` to `<cmath>`. See https://github.com/Qucs/qucs/issues/154 for additional information. It builds OK on Linux and `gcc-4.8.2`. Maybe your MinGW version supports <cmath> only partially. Unfortunatelly, I have no Windows and no MinGW and I cannot test and fix it. Maybe add `#ifdef` directive for `M_PI` for Windows?

guitorri, 2015-01-10:
`M_PI` is not C++ standard. Perhaps MinGW is being pick and sticking to C++.
See http://www.cplusplus.com/forum/beginner/105176/
Either a `#ifdef` or a constant should help, perhaps we can add them to the `config.h` ?. `M_PI_2` and [other defines](http://www.gnu.org/software/libc/manual/html_node/Mathematical-Constants.html) should be fixed as well.

ra3xdh, 2015-01-10:
In this case we have no choice. We should put some `#ifdef` for `Q_OS_WIN` or something else in `config.h`.

in3otd, 2015-01-10:
yes, it's strange, I used MinGW [i686-4.8.2-release-posix-dwarf-rt_v3-rev3.7z](http://sourceforge.net/projects/mingw-w64/files/Toolchains%20targetting%20Win32/Personal%20Builds/mingw-builds/4.8.2/threads-posix/dwarf/i686-4.8.2-release-posix-dwarf-rt_v3-rev3.7z/download) , as suggested on the Wiki (@guitorri, do we have a choice or are we tied to it since Qt was compiled with it?) , but on linux with `gcc 4.8.2` also here it compiles fine (and also `clang 3.3`).
Note that of course `M_PI` and similar are used in several other places in the code (and often defined in a local header)

guitorri, 2015-01-10:
@in3otd , the fastest way to get Qt is to install the official SDK binaries. To avoid headache it is best to use the compiler they recommend. The Qt installer even check for the (MinGW) compiler. Unless you want to build Qt from source on Windows... I prefer and also suggested the shortest route on the Wiki.
I have the impression that MinGW headers are not the same as the Linux GCC.

guitorri, 2015-01-13:
The brute force method used in a few places (diagramdialog.h, main.h, qucs-transcalc/units.h qucs-filter/qf_cauer.cpp) is to simply add the needed constants.

```
#ifndef M_PI
#define M_PI 3.14159265358979323846
#endif
```

This has the drawback of not being type checked.

Perhaps we can add the constant either to the `main.[h,cpp]` or `config.h`

```
const double M_PI = 3.14159265359;
```

or

```
namespace math
{
    const double M_PI = 3.14159265359;
}
```

When I asked on another issue about using more compile flags, perhaps the `-ansi -pedantic -Wall` would help us to clean up a bit the rust.

in3otd, 2015-01-13:
I didn't have time to look much into this, I googled around and as usual there a lot of different opinions and solutions.
I thought to simply add a check in `configure.ac` so that files needing these constant will just need to include `configure.h`, but then a similar solution will need to be done for CMake, and I am not familiar with it.
Regarding type checking, can it really be a problem with a simple object-like macro like this??

Speaking of compiler flags (maybe I should open another issue for this), what about defining `QT3_SUPPORT WARNINGS` so we can more systematically remove also the small Qt3-related items ?

guitorri, 2015-01-14:
On `qucsator` we have no alternative. I will use the `namespace` with `cons [type]` option to scope the constants and avoid collision with other linked modules. For consistency I suggest we do the same for the rest of the package. I suggest we use `const [type]` as needed. The namespaces are not yet in use in qucs-gui.

I did not know about QT3_SUPPORT WARNINGS. Up to know I changed the Qt3 header and followed the compiler error trail. Please enable this warning. It will make us all more aware of what needs to be done.

guitorri, 2015-01-17:
I will take a look at this one.

guitorri, 2015-01-20:
Should be fixed with merged in 6c1fb1e964c3a9199e7ba433fc27aa8fabb16025

closed 2015-01-20 by guitorri