Qucs/qucs #177: Add namespace protection on misc.h

yodalee, 2015-01-13:
There are some rarely use function in misc.h, their definition are exposed to the global space.
We need add a namespace protection (something like namespace Misc)

guitorri, 2015-01-16:
This a good point. If we are going to start using namespaces we have to do it thoroughly. Perhaps we can wait till we move past Qt3Support?

yodalee, 2015-01-16:
I think this can be done in parallel. The work is simply add namespace in header file, and add namespace at every function call. An easy but tedious work orz.

yodalee, 2015-01-17:
I will do this (writing a script =w=)

guitorri, 2015-01-17:
The [Doxygen docs](https://github.com/Qucs/qucs-doxygen) might be helpful on identifying the hierarchy.
What are your thoughs about the names of the namespaces?

yodalee, 2015-01-17:
I'm not sure, maybe just `misc`, since it's just a secure to prevent pollution from other file.
I complete the script:
```
for i in complexRect complexDeg complexRad StringNum str2num num2str \
StringNiceNum convert2Unicode convert2ASCII properName properAbsFileName \
properFileName VHDL_Time VHDL_Delay Verilog_Time Verilog_Delay Verilog_Param \
checkVersion; do  
  find . -name "*.cpp" -print0 | xargs -0 sed -i "s/$i/misc\:\:$i/g" 
done 
```

closed 2015-01-19 by yodalee