Qucs/qucs #181: Xyce and Qucs integration

ra3xdh, 2015-01-14:
Xyce is a new spice-compatible simulation kernel. It provides circuits simulation using parallel computations (via openMPI ). More info about Xyce at http://xyce.sandia.gov/. The purpose of Qucs<-->Xyce integration is simulation of Qucs schematics with Xyce. 

Here is tasklist for Xyce<-->Qucs integration
 - [ ] Xyce-compatible netlist builder
 - [X] Xyce output results parser and Xyce output-->Qucs dataset converter
 - [ ] basic electronics components (RCL and semiconductor) support
 - [ ] file componets support
 - [ ] Qucs library components support
 - [ ] Qucs equations to Xyce equations translation
 - [ ] `Spice Equation` component
 - [ ] Basic simulations (AC,DC,Tran) support
 - [ ] Parameter sweep support
 - [ ] Harmonic balance support

ra3xdh, 2015-01-17:
Simulated the first circuit with Xyce and Qucs! I have added combo box for simulation backend selection. Check `Xyce (Serial)` and then press `Simulate`. Then go to display page, put diagram, and select `<circuit_name>_xyce` dataset. Source code available at https://github.com/ra3xdh/qucs/tree/spice4qucs . Screenshots are here.
[image]

[image]

guitorri, 2015-01-17:
Terrific!! :clap:

guitorri, 2015-01-17:
I can't wait to start comparing the accuracy of `qucsator`against the other kernels... Are you keeping an updated list of components that you already ported (ngspice and xyce) somewhere?

Perhaps we could think of a component filter based on list of supported components. I mean, to help the user know which components are currently handled for npspice and xyce. So, the UI adapts to the backend. Just thinking...

in3otd, 2015-01-17:
Excellent work @ra3xdh ! 

@guitorri, I see some more work for you in the documentation stuff here :smile: ... I was already thinking that the components documentation done via Doxygen needs an additional section saying which simulations are supported (e.g. HB is often not supported, S-param. file does not support TR, etc.)... now need to add also which *simulators* are supported...

ra3xdh, 2015-01-17:
@guitorri  The list of supported components is on Wikipage here https://github.com/Qucs/qucs/wiki/QEP:-Qucs-schematic-simulation-with-ngspice . Xyce supports all of Ngspice supported components except semiconductor devices (diodes, transistors). I will fix semiconductor devices support with Xyce during the next week. There is a list of supported simulations on this wikipage.

ra3xdh, 2015-01-17:
I compared `qucsator` and `ngspice` precision. For my test circuit there was a little divergence between `ngspice` and `qucsator` for transient simulation. Something like this. 
[image]

ra3xdh, 2015-01-18:
Fixed diodes and transistors support for Xyce. For example audio amplifier test circuit was simulated. This schematic could not be simulated with `qucsator`.
[image]
[image]

ra3xdh, 2015-01-20:
Harmonic balance support added. Here is example of simulated circuit. Node names conversion was required to display HB results correctly. There are both results from Qucs and from Xyce on this screenshots.
[image]

guitorri, 2015-01-20:
Very nice!

MikeBrinson, 2015-01-21:
Thanks Vadim
This is an important step forward.
Using Xyce Harmonic Balance simulation will allow
Qucs HB models to be checked for function and accuracy.

ra3xdh, 2015-02-14:
Added basic EDD support via B-type sources. Only currents are supported, charges aren't supported. Now in's need to implement variables support and Qucs equations to spice equations translator. As alternate we can implement `Spice Equation` component. B-voltage and -currrent sources also should be added. These components will be available only for spice simulations.
[image]

ra3xdh, 2015-02-15:
I tried to add EDD charge equations support via current source I=dQ(V)/dt . It leads to I=ddt(Q(V)) B-type current source. This source cannot be simulated with Xyce. It fails to calculate initial DC conditions. Ngspice has no diff() function or equivalent. It seems to be unable to simulate charge-oriented EDD with Ngspice. 

Here is converted Xyce Netlist (nonlinear capacitance example from equations tutorial):
http://pastebin.com/7aRJhZ3D

Here is simulation log:
http://pastebin.com/9ihGZucg

MikeBrinson, 2015-02-15:
Hello Vadim
Thanks for the charge equation data.I will have a look at it tomorrow and send you my findings.
Best wishes.
Mike
 Mike Brinson

mbrin72043@yahoo.co.uk

   

http://pastebin.com/7aRJhZ3DHere is simulation log:
http://pastebin.com/9ihGZucg—

MikeBrinson, 2015-02-16:
Hello Vadim
I have found a solution to Qucs/ngspice/Xyce  EDD charge problem.
I agree with you that Xyce function ddt does not seem to work as expectedand indeed in a similar way to Verilog-A.
Have a look at the attached pdf - this shows how to use the B type I = .....source to generate dQ/dt as the voltage across a 1 Henry inductance.In the example I have just shown a linear capacitor.  It is of course possibleto make this non-linear by changing the B I source equation.
I also attach a basic SPICE netlist for the same test circuit.  This runs OK with 
Xyce.
One advantage of using the approach show in the pdf is that it will work with bothngspice and Xyce.
I read your latest post concerning Qucs probes and their use with ngspice and Xycewith interest.  Regarding the Qucs current and differential voltage probes.   These work withall Qucs simulations except Harmonic Balance.   Would it be possible for you to add the code 
so that they work with Harmonic Balance.  This would be an important step forward for Qucsand would make testing Xyce HB more straightforward.
If you need more information on the above please email me.
best wishes.
Mike  

 Mike Brinson

mbrin72043@yahoo.co.uk

   

http://pastebin.com/9ihGZucg—

ra3xdh, 2015-02-16:
@MikeBrinson , attachment was lost on Github. Please send it via e-mail to me directly.

MikeBrinson, 2015-02-16:
Hello Vadim
Find attached lost pdf.
Mke Mike Brinson

mbrin72043@yahoo.co.uk

ra3xdh, 2015-02-16:
Fixed HB support for I and V probes
[image]

ra3xdh, 2015-02-17:
Added charge equations support for both Xyce and Ngspice. Now EDD is fully supported.
[image]

ra3xdh, 2015-03-21:
Added first working parameter sweep support (only Xyce, Ngspice will be added later). Unfortunately, But transient simulation with par sweep don't work.  Qucs cannot support datasets with adaptive step. Xyce and Ngspice use adaptive time step and time step value for each parameter sweep differs from specified time step. Qucsator uses the same time step for each parameter sweep step.

There may be two ways to overcome this issue:
1. Add support of adaptive time-steps for qucs dataset. Qucs datasets may become incompatible with datasets from old Qucs
2. Redistribute transient simulation data form Xyce/Ngspice using linear interpolation to obtain equal time step and simulation points count for each sweep varaible step.

What is your opinion?

Here is example of AC-simulation with parameter sweep.
[image]

ra3xdh, 2015-03-24:
I have added initial support of Parameter Sweep for Ngspice. It seems Ngspice has no problems with adaptive time step at parameter sweep.
[image]

felix-salfelder, 2015-07-20:
may i suggest to remove the checkmarks in the first post and insert them, once the features have been really merged? if there are related commits, it would be nice to add some references.

ra3xdh, 2016-06-24:
Added new "XYCE script" simulation type. It allows to place on schematic XYCE code that will be directly passed to simulation kernel. This simulation type allows to get full access to all features of XYCE. Multiple XYCE scripts in one schematic are supported. Every script represents one simulation. Attached example illustrates this new simulation type. AC and TRAN analysis of BJT amplifier is performed. Phase response is obtained via XYCE `VP()` statement. Postprocessing with `.MEASURE` is used for TRAN.
[image]

ra3xdh, 2017-06-29:
Closing, because #275 is closed. See details at #275 (comment) Qucs-S and XYCE support development will be continued at http://ra3xdh.github.io/ Use issue tacker https://github.com/ra3xdh/qucs/issues to report bugs on spice4qucs.

closed 2017-06-29 by ra3xdh