Qucs/qucs #182: Qucs (GUI) translation workflow.

guitorri, 2015-01-16:
I mentioned already in #179 that I am setting up a Qucs organization on Transifex to manage the translation of Qucs. Transifex is crowd-sourced platform for translation.

I am still testing and checking how the merges happen when pulling back the translations from there.

The page is up in:  https://www.transifex.com/projects/p/qucs-desktop/

I will invite (with priority) translators that contributed in the past to become the _language coordinator_. Later I will open to anyone, first come first serve.

I also subscribed to the Microsoft Bing Translator to provide Machine Translation on the Transifex webapp (one has to pay for the Google Translator API key). It is quite handy and speedup the process. With one button the string is translated, you check it for typos and save.

One thing that I observed it that we can drop the `qucs/translations/qtgeneric_<lang>.ts` and use instead the `qt/4.8.6/translations/qt_<lang>.qm` shipped with Qt. The only language that we might want to keep   is Turkish (tr) which is available on Qt 4.8.6.

For the above reason only the `translations/qucs_<lang>.ts` were pushed to Transifex.

This is a list of changes:
- [ ] either import the `qt_<lang>.qm` into the tree, or copy them into the package, or install in case of source build
- [ ] modify the QTranslator to load both `qt_<lang>.qm` and `qucs_<lang>.qm`. See [main.cpp](https://github.com/Qucs/qucs/blob/master/qucs/qucs/main.cpp#L474-L479)
- [ ] update the settings. See [qucssettingsdialog.cpp](https://github.com/Qucs/qucs/blob/master/qucs/qucs/dialogs/qucssettingsdialog.cpp#L91-L114)
- [ ] switch languages at runtime (this is optional). See [qt-project.org example](http://qt-project.org/wiki/How_to_create_a_multi_language_application)

in3otd, 2015-01-16:
Good work, Guilherme!
There are a few (very few, actually) typos in the original translatable strings, will it be better to correct those now? We could correct both the source strings and the strings in the `.ts` files, so we won't lose any translated string.
Those are obvious things like
*relais* -> *relays*
*verlog* -> *Verilog*
*Grey code* -> *Gray code*
*frist* -> *first*

There are other things that are not really wrong but should maybe written differently:
- AmE vs BrE, like *modeling* -> *modelling* or *color* -> *colour* . Since we have a `qucs_uk.ts` file, I guess we should use the American English form
- consistent style for acronyms, we have *diac*, *triac* and *SCR* ; *flip flop* and *Flip Flop* (but not *flip-flop*!) and the variants *jk flip flop* and *JK-FlipFlop* 

...but probably we will need a  *main language coordinator* (or worse, a committee :grin:) to decide on these latter...

guitorri, 2015-01-16:
@in3otd, I have to read the docs about how to handle the merges. I did with PO files, but TS files are giving a bit of difference and might need a bit of scripting to set things straight. 

We will have to learn to handle any situation and setup the guidelines that make the work less enjoyable. Please go ahead ad fix the typos if you like. I will figure out a way to sync them with Transifex.

The motto is to keep things simple. For now I would say that we open the TS in a text editor and and set the spellchecker to en_US and then en_UK. The one that has less changes is the default English. The if the community wands to localize the other we can always do that.

yodalee, 2015-04-15:
I think this can write into wiki too.

guitorri, 2015-09-17:
Some changes mentioned in this issue are already merged with PR #357.
I will change the milestone so we can continue on this later on...