Qucs/qucs #185: Option to redirect `log.txt` and `netlist.txt`

guitorri, 2015-01-17:
Currently whenever a simulation is run the  `log.txt` and `netlist.txt` files are (over)written on the QucsHomeDir (typically `~/.qucs`). 

This is annoying especially if a user wants to compare the netlists of two similar circuits. 

These are the situations I could think of:

- Default to QucsHomeDir (typically `~/.qucs/`, possible to change on the settings)
- [ ] add option to redirect the files to the project directory. (ex `~/.qucs/[project]_prj/`)
- [ ] add option for a temporary/scratch directory to keep intermediate files (ex `~/temp/[project]_temp/`)

Are there any other thoughts on how we could handle this?

ra3xdh, 2015-01-17:
Maybe add  `Build netlist` menu entry to `Simulate` menu or something similar? Save dialog for netlist file could be shown by clink on such menu item.

yodalee, 2015-01-17:
redirect file to `~/.qucs/[project]_prj/` won't solve the problem. Since every simulation in this project will replace each other, and usually circuits user want to compare exist in same project.
I think maybe we generate the netlist file with circuit name inside, such as netlist_[circuit].txt.
And log.txt, maybe we can use append, not over-write the file.

guitorri, 2015-01-17:
@ra3xdh, updated on the checklist. An option to build only the netlist without running the solver is also nice to have.

@yodalee, you are right. One project may contain several schematics. If the user chooses to redirect we can append the netlist as you propose. I strongly believe that one a simulation run should have a separate log and netlist. To make things easy to reproduce. We can thing about a separate log for the duration of the session.

in3otd, 2015-01-17:
Agree that the `Build netlist` menu entry will be nice to have. Saving the netlist, could be done thru the editor used to show the netlist. Currently `qucsedit` does not have a `Save as...` but could be added. So actually we will need a menu `Build and show netlist`, then the user can decide whether to save to another file or just read it on the screen.
Similar for the log messages, they can be already seen on screen using `Show last messages`, if `qucsedit` will have a `Save as...` that could be enough. In my experience one does not compare netlists and logs very often, even if having the possibility to save them to a separate file within Qucs is certainly useful.
And yes, `log.txt` and `netlist.txt` should probably be in the project directory (or the file current directory, if not belonging to a project), not always in QucsHomeDir .
Additional question, should we delete those files on exit? Currently if you do a `Show netlist` or `Show last messages` on a newly open project you are shown the content of the existing files, might be slightly confusing.

yodalee, 2015-01-20:
There is another issue related to this: current version qucs allow user to open a schematic file, save it anywhere, simulated it directly. If we need the netlist and log files to be store at project directory, we must limit that files can only be open after use open a schematic.
Actually I prefer this. This makes sure that every files in open in a Project, eliminating the exception in implementation of project file manager.

in3otd, 2015-01-20:
yes, but... I often need to quickly test a schematic, for circuits which I do not plan to develop further after one or two simulations and in this case **not** being forced to create a project is a good thing. I already have too many projects to scroll over in the `Projects` tab...