Qucs/qucs #188: appVeyor seems having some error

yodalee, 2015-01-18:
I just push some new commit to the PR branch, and triggering Travis-CI and new CI appVeyor. The latter seems having an error on command line.
it clone the project with address:
`git clone -q git://github.com/Qucs/qucs.git C:\projects\qucs`
which should be:
`git@github.com:qucs/qucs.git`
******
Also there is a redefinition in the qucs-core code.
https://travis-ci.org/Qucs/qucs/jobs/47417012

guitorri, 2015-01-18:
Sorry appVeyor was active by mistake... I am still experimenting to see how it works.

guitorri, 2015-01-18:
I delete the AppVeyor project for now. I still have to figure out how to install Qt4 in there. They seem to have Qt5 only.

@yodalee, what do you mean by redefinition, that link is from a branch that Bastien is working on...

yodalee, 2015-01-18:
My fault, no qucs-core error.

closed 2015-01-18 by yodalee