Qucs/qucs #193: Google Summer of Code 2015

ra3xdh, 2015-01-22:
Guilherme and I find useful for Qucs to take a part at Google Summer of Code 2015 https://www.google-melange.com/gsoc/homepage/google/gsoc2015 . Here you can discuss this point. What is your opinion on GSOC-2015? Should we take a part here?

Here may be a list of our ideas for GSOC-2015 https://github.com/ra3xdh/qucs/wiki/Qucs-GSOC-2015. If you find GSOC-2015 useful for us I will move it to main Qucs Wiki. You can add or remove ideas from this ideas page. 

If we find GSOC-2015 useful, we need to elect our administrator for contacts with Google and mentors. I can serve as mentor for my ideas. Administrator's functions are undesired for me because I am busy now with Qucs/Spice interfacing development.

ra3xdh, 2015-02-08:
There is no comments on GSOC-2015. It seems nobody finds GSOC-2015 harmful for Qucs. So I will continue with our proposal for GSOC-2015. I put a draft of proposal and ideas list at github repo https://github.com/ra3xdh/qucs-gsoc2015 . You can send me pull request if you have suggestions or your own ideas. Please read carefully sections concerning mentors, students, and ideas list.

Google needs Administrator from our organization. Administrator will be responsible for contacts with Google. Guilherme cannot be our Administrator at GSOC-2015. There was no people that wish to be GSOC-admin. For this reason I will be GSOC-2015 admin. If you have another candidate you can propose him at comments here. You can also propose your own mentor candidates.

timofonic, 2015-02-08:
Hello.

I'm just a novice, but I would love if there could be an easy integration with KiCAD. CERN proposed it, so maybe it could even be sponsorized as an extra-slot project or whatever.

Why simulators and schematic+PCB software need separate libraries? What about merge them? Or at least make it automatic. 

I'm being teached electronics by a combination of Eagle and Proteus. It's painful to need to redraw the same schematic again and again, Proteus has lots of simulation flaws too.

ra3xdh, 2015-02-08:
I have mixed feelings on Simulator-->PCB editor link. Currently I am lecturer at University. Previously I worked during seven years as electronics engineer.  My point of view is based on my experience as designer of electronic equipment. Not all components on PCB could be simulated (for example Microcontrollers (MCU), Complex ICs, FPGAs, etc.) and these components ofted also should not be simulated. So it will require exclude them from simulation. It's better split your device to subcircuits and simulate them. Whole circuit on PCB requires a long time to simulate it and loads computer. So it is often unable to transfer the whole netlist from simulator to PCB. Only a very simple circuits could be fully simulated. Other circuits requires splitting. And I don't see any useful in integration of PCB editor in Qucs. Deep integration with PCB design tools is also useless. It is just my opinion based on my experience.

You mentioned Proteus. I used it in production. It allows to simulate only a very limited functionality of circuits. Many MCU circuits work fine on physical board and cannot be simulated with Proteus . Only AVR support in Proteus seems fine.

I suggest the following way. We can include `KiCAD symbol` or `KiCAD footprint` sections in our component library or some reference or KiCAD library. Also we can add `Package` (for example TO-220, TO-95, TO-5, etc.) property for single not-library components. Also we need to implement `Export to KiCAD netlist` feature. This feature will be useful. You can prepare your proposal. There was some similar projects on Github https://github.com/Valber/qucs2kicad. This project seems to be abandoned but you can use it as startpoint.

timofonic, 2015-02-08:
I'm a novice electronics student, I was unable to learn computer programming. I was suggesting a proposal for some interested third party, but I'm just a n00b that wants to replace Eagle+Proteus with a saner and open source environment.

About circuit simulation: So what about Altium and others? Those EDA suites feature big support of MCUs and FPGAs.

I'm just a student. I "just" want accurate (more than Proteus, please) and useful simulation of circuits (if possible, able to select between and theorical simulation) to learn from it (and check exercises, theory...) and the most difficult stuff we are going to use are AVR, despite I have tons of curiosity to learn more.

I see a "simulate" button on EDA as a dream to happen, because the faster and easier is to simulate the earlier I can play with the simulation and take measurings with virtual instruments. That's one of the few things I liked about Proteus.

Maybe beginners have different needs and workflows than professionals in a production environment.

ra3xdh, 2015-02-20:
I registered Qucs for GSOC-2015. I assigned yourself as administrator. Now we need one person as backup administrator. This person should register at https://www.google-melange.com/gsoc/homepage/google/gsoc2015 till 20 February 7:00PM (UTC). I we cannot provide backup administrator, our application will be rejected.

ra3xdh, 2015-02-26:
Now our application process is completed. We will see result on March 2.

ra3xdh, 2015-03-03:
Our application for GSOC-2015 was rejected.

closed 2015-03-03 by ra3xdh