Qucs/qucs #196: SubWindows support

arhiv6, 2015-01-25:
Hi All. I started adding support subwindows into Qucs main window. I replaced QTabWidget *DocumentTab on QMdiArea *DocumentMdi. The default use classical TabView, but you can switch to SubWindowsView (use menu View). Now the application is not stable (I have not yet completed the and I have a little experience). Should I continue to do so now?  See branch https://github.com/arhiv6/qucs/tree/subWindows  
[image]
[image]

ra3xdh, 2015-01-26:
I have built and test it. It doesn't break old interface. User can select what interface he wants. AWR has similar tile-windowed interface. It may be useful for big monitors. For small laptop monitors it is useless. Tile windows become too small. Tile windows could be useful for large monitors. Cascade windows concept duplicates tab concept. I think cascade windows are useless. Please inform people at Qucs-devel mailing lists about your concept to get further feedback.

I found a bug. Qucs crashes when I try to open file using File->Open.

yodalee, 2015-01-26:
May I ask that, is AWR like Qucs, using Tabwidget to manage all its document: schematic, simulation result, Verilog document, EM simulation?

guitorri, 2015-01-28:
This is interesting. @yodalee Google images suggest that AWR can have both, tabbed **and** sub-windows. I guess you can draw and drop tabs/sub-windows around. Other commercial tools have only tabbed **or** multi-window. On Virtuoso when you go down into a sub-circuit you an either open it in a tab or on a new window (also with tabs).
We could also think of having multiple views of the same schematic. 
There are many options. We should think a bit and **design** the interface for usability... 
This might help us with the terminology [wiki Multiple_document_interface](http://en.wikipedia.org/wiki/Multiple_document_interface)

yodalee, 2015-01-28:
That is funny, I have to think about how they realize it (Though I think multiple windows doesn't lose any usability)

in3otd, 2015-01-28:
there are a few videos where you can see AWR with both tabs and subwindows, e.g. [here] (https://www.youtube.com/watch?v=2MdhWdLFcz8) . 
IMHO, multiple separate windows not as practical to switch between them when you have many open at the same time, but well, if we can have the choice of having both solutions :wink: ...

guitorri, 2015-01-30:
Related: https://sourceforge.net/p/qucs/feature-requests/57/

arhiv6, 2015-01-31:
I create QMdiArea inheritor (file qucsmdiarea.h and qucsmdiarea.cpp), changes in the rest of the code are minimal. Now the program works, but has some problems. The reasons for this: I have a little experience + QMdiArea in Qt (I use 4.8 on Ubuntu) has bugs, especially in TabView mode (for example see [QTBUG-44109](https://bugreports.qt.io/browse/QTBUG-44109)). Can you tell me that made bad in my code, and does my code suitable for the project?

closed 2018-07-19 by guitorri