Qucs/qucs #200: onGrid Action seems useless?

yodalee, 2015-01-28:
The Action in `positioning->set on grid` seems no work?
Since everything on schematic is always on grid, actually I cannot find a way the make item off grid.
I think making every items always on grid is a better design. Off grid only brings chaos to user. For example, users move a line but it just cannot connect it to node because of off grid property.
I suggest that Schematic item always on grid and remove the "set on grid" button.

in3otd, 2015-01-28:
well, of course schematic items should be on grid always, but if you try hard enough you can force them off grid. Take for example a `Bias T` and rotate it several times (keep `Ctrl-R` pressed for a couple of seconds); you will see that the components slowly walks around and is no longer aligned to the grid. The only way to realign it is with the 'set on grid' command. This problem is more evident for paintings, but of course less important there.
I put a `\todo` line in `schematic.cpp` some time ago about this issue, but then never found the time to work on this, also because I feel is quite low priority.
So, it could be removed from the right-click component menu if you wish, since it is not really often used, but should remain available in the menu just in case the user brings thing off grid by mistake.

guitorri, 2015-01-28:
It kind of make sense to have everything on grid by default. But then we have to check all components and ports for rotation and mirror operations to make sure the nodes always fall into a grid point. Otherwise the wire will have to go off grid to make the connection... What if the user changes the grid size? Perhaps we can work it out along with the QGraphics port. I have done little some progress on my fork. Soon I will put an issue to track the port. It is a lot of work, but things are seeming to fall into place.

yodalee, 2015-01-29:
Oh, that is funny, well maybe we can try to solve the `rotate` issue in new Schematic editor. I didn't notice that.
I think we can set grid size constant in schematic editor actually, my experience on Agilent ADS is that I seldom need to adjust grid size in Schematic editor.

closed 2015-01-31 by yodalee