Qucs/qucs #201: Regresion on the 555 example

guitorri, 2015-01-30:
These example works fine on 0.0.18.
 - [555_examples_prj](https://dl.dropboxusercontent.com/u/29028094/qucs/555_examples_prj.zip)

The two top schematics cause segfaults with current master.

Traceback:

```
NOTIFY: TR1: solving transient netlist
Process 89081 stopped
* thread #1: tid = 0x56f0d9, 0x000000010016df53 libqucs.0.dylib`qucs::history::seek(this=0x0000000100e09cb8, tval=-0.0000000050000000000000001, l=0, r=-1, diff=0x00007fff5fbfd758, idx=-1) + 131 at history.cpp:129, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x000000010016df53 libqucs.0.dylib`qucs::history::seek(this=0x0000000100e09cb8, tval=-0.0000000050000000000000001, l=0, r=-1, diff=0x00007fff5fbfd758, idx=-1) + 131 at history.cpp:129
   126 	  int i = (l + r) / 2;
   127 	  if (l == r)
   128 	    return i;
-> 129 	  nr_double_t v = (*this->t)[i];
   130 	  nr_double_t d = v - tval;
   131 	  if (fabs (d) < diff) {
   132 	    // better approximation
(lldb) bt
* thread #1: tid = 0x56f0d9, 0x000000010016df53 libqucs.0.dylib`qucs::history::seek(this=0x0000000100e09cb8, tval=-0.0000000050000000000000001, l=0, r=-1, diff=0x00007fff5fbfd758, idx=-1) + 131 at history.cpp:129, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  * frame #0: 0x000000010016df53 libqucs.0.dylib`qucs::history::seek(this=0x0000000100e09cb8, tval=-0.0000000050000000000000001, l=0, r=-1, diff=0x00007fff5fbfd758, idx=-1) + 131 at history.cpp:129
    frame #1: 0x000000010016de4f libqucs.0.dylib`qucs::history::nearest(this=0x0000000100e09cb8, tval=-0.0000000050000000000000001, interpolate=true) + 239 at history.cpp:113
    frame #2: 0x0000000100079d6e libqucs.0.dylib`qucs::circuit::getV(this=0x0000000100e19250, port=1, t=-0.0000000050000000000000001) + 62 at circuit.cpp:940
    frame #3: 0x00000001002962c7 libqucs.0.dylib`digital::getVin(this=0x0000000100e19250, input=0) + 71 at digital.cpp:76
    frame #4: 0x00000001002964eb libqucs.0.dylib`digital::calcTransferX(this=0x0000000100e19250, input=0) + 459 at digital.cpp:86
    frame #5: 0x000000010029658b libqucs.0.dylib`digital::calcTransfer(this=0x0000000100e19250, input=0) + 27 at digital.cpp:91
    frame #6: 0x0000000100297b9e libqucs.0.dylib`logicnor::calcOutput(this=0x0000000100e19250) + 318 at nor.cpp:45
    frame #7: 0x0000000100297635 libqucs.0.dylib`digital::calcTR(this=0x0000000100e19250, t=0) + 325 at digital.cpp:180
    frame #8: 0x0000000100150093 libqucs.0.dylib`qucs::trsolver::calcTR(self=0x0000000100e16900) + 83 at trsolver.cpp:740
    frame #9: 0x00000001000b47e3 libqucs.0.dylib`qucs::nasolver<double>::calculate(this=0x0000000100e16900) + 51 at nasolver.h:72
    frame #10: 0x00000001000b2c4f libqucs.0.dylib`qucs::nasolver<double>::solve_once(this=0x0000000100e16900) + 47 at nasolver.cpp:143
    frame #11: 0x00000001000ae45f libqucs.0.dylib`qucs::nasolver<double>::solve_nonlinear(this=0x0000000100e16900) + 1263 at nasolver.cpp:480
    frame #12: 0x000000010015050c libqucs.0.dylib`qucs::trsolver::corrector(this=0x0000000100e16900) + 28 at trsolver.cpp:574
    frame #13: 0x000000010014ec1a libqucs.0.dylib`qucs::trsolver::solve(this=0x0000000100e16900) + 2138 at trsolver.cpp:281
    frame #14: 0x000000010009235c libqucs.0.dylib`qucs::net::runAnalysis(this=0x0000000100e08190, err=0x00007fff5fbfee24) + 1628 at net.cpp:264
    frame #15: 0x0000000100003a2d qucsator`main(argc=5, argv=0x00007fff5fbff9b8) + 8957 at ucs.cpp:251
    frame #16: 0x00007fff81edf5fd libdyld.dylib`start + 1
    frame #17: 0x00007fff81edf5fd libdyld.dylib`start + 1
(lldb) 
```

guitorri, 2015-01-30:
@bastien-roucaries would take a look at this?

bastien-roucaries, 2015-01-31:
Could we upack example on git tree ? I will allow to do regression testing
on it.

Bastien

in3otd, 2015-01-31:
...since it seems we will soon start some work on the transient solver, it will make sense to add several known currently-failing circuits to the test suite. Maybe leave them disabled by default and add a switch to include also (or only) them for testing during the development. I have already collected a few from the various issues and discussions, will later post a message to `qucs-devel` to ask for some more examples which fail the transient simulations

guitorri, 2015-02-09:
There are some numerical differences with respect to 0.0.17 and 0.0.18. 
I am trying to bisect it.

guitorri, 2015-02-09:
Sorry for the noise. The issue is not with the 555 example...

commit 4244e919 (2015-01-31): Avoid a signed integer comparaison versus unsigned integer

Close #201
closed 2015-02-09 by guitorri