Qucs/qucs #204: Change Windows Environment Variable `HOME` to something else (like `QUCSHOME`)

jrast, 2015-02-20:
With the current setup, using `HOME` to define the Project and Settings directory, Qucs is causing problems as other programs might also look for a environment variable called `HOME`. In my case this is cygwin wich relies on the HOME Variable, but there might be other tools rely on this variable (which is also not nice from these tools to use generic variable like HOME). 

My recomondation ist to change this variable to something more specific like `QUCSHOME`.

guitorri, 2015-02-20:
I think HOME is not used anymore and can be dropped from the Inno instaler script.

These are the environment variables that we currently use:
```
./qucs/main.cpp:641:  char* var = getenv("QUCSDIR");
./qucs/main.cpp:672:  var = getenv("ADMSXMLBINDIR");
./qucs/main.cpp:689:  var = getenv("ASCOBINDIR");
./qucs/main.cpp:706:  var = getenv("OCTAVEBINDIR");
```

guitorri, 2015-05-23:
Should be fixed with #255

closed 2015-05-23 by guitorri