Qucs/qucs #205: Reminder about qucs-test, platform dependent issues and numerical differences.

guitorri, 2015-02-21:
This issue is a reminder about the numerical issues mentioned below. 

This issue should cleared/understood before the next release.

Tests failing the numerical check are being skipped (on [Linux](https://github.com/Qucs/qucs-test/blob/master/skip.txt) and [OSX](https://github.com/Qucs/qucs-test/blob/master/skip_OSX.txt)) to allow Travis CI to finish with success.

Running the [qucs-test](https://github.com/Qucs/qucs-test) on different platforms may lead to numerical differences (with respect to results generated in qucs 0.0.17, linux 64).

See these commits with figures for Darwin, Windows and Linux (see branch name):
https://github.com/Qucs/qucs-test/commit/aee5f5a5809a6abaecc07a7fa1725b25e768ce32
https://github.com/Qucs/qucs-test/commit/e86c82ec4da1215593aa40d4f2a155d99577558e
https://github.com/Qucs/qucs-test/commit/8941f2b8e93b78bdb21d2529a21133594f89277a

The above tests were executed with rtol=0.1 and atol=1e-5.
The numpy.allclose returns true if the absolute error of two arrays is lower-equal than the sum of two tolerances. Like: abs(a-b) <= (atol + rtol*b)

Looking at the pictures more carefully the results are not that far apart.
Indeed fast switching and accumulated error seem to be related to these issues.

On TR_Puls3b_prj the tswitch seems to be toggled one step before/after of what used to be.
On Windows (32bit !!) some circuits have a phase reversal at start and the accumulated error is much more pronounced.

yodalee, 2015-04-15:
@guitorri maybe you can write this into wiki?

guitorri, 2015-04-16:
Write what in the wiki?? 
Platform dependency is an issue. So far we could not trace the root of the problem. We need more examples and cross-check with other simulators to see if we can pinpoint the source of the differences.

yodalee, 2015-04-16:
I think I misunderstand the issue, sorry about that.

guitorri, 2016-11-30:
Reminder to check what happened in #623

guitorri, 2017-10-17:
Some reference files still show differences after #206.