Qucs/qucs #226: Prepare spice4qucs for merge

ra3xdh, 2015-04-18:
Spice4qucs is a faeture that enables to simulate Qucs analog circuits with Ngspice and Xyce engines. See #77 and #181 discussion thread for more info. I and Mike Brinson made more than 80 commits at https://github.com/Qucs/qucs/tree/spice4qucs during more than 6 months. Spice4qucs sufficiently extents power of Qucs. I think that it's time to merge it in master. We need to solve the following tasks:

- [ ] Add EDD support in qucs2spice netlist converter. I am currently working on it. It can be done. There is nothing complex;
- [ ] Add simulator (ngspice, runxyce) executables location settings in Simulation dialog. It will allow to use simulators that are not in $PATH;
- [ ] Add a button "Save spice netlist as..." in simulation dialog;
- [ ] Add a simple tutorial on basic simulations (AC,DC,TRAN) and equations usage with Qucs and Ngspice/Xyce;
- [ ] Add schematic examples with Ngspice and Xyce simulations in examples subdirectory. We can use existing examples from spice4qucs discussion thread. 
- [ ] Rebase to master

guitorri, 2015-04-18:
See #227 for Autotools.

There are a few warning of unused variables (`isXyce`). 
Can you add some Doxygen documentation?
I just build it, I will try to find some time to _use it_ before I give some more feedback.

ra3xdh, 2015-04-18:
Thank you very much for #227! Remaining "unused `isXyce` variable" warnings concerns components that have the same netlist entry for both Ngspice and Xyce. For these components `isXyce` parameter remains unused. I will add  Doxygen task to tasklist.

guitorri, 2015-04-18:
You can use default values or `Q_UNUSED()` macro to silence the warnings...

ra3xdh, 2015-04-18:
@guitorri , See last commit on upstream `spice4qucs` . Warnings are fixed.

ra3xdh, 2015-04-19:
Implemented EDD in Library components support.
[image]

ra3xdh, 2015-04-22:
Added simulators executable setup feature. It is implemented through special new dialog `simsettingsdialog.cpp` . It is especially important for Windows Qucs users. Also added simulator selection combo box in diagram dialog. 
[image]
[image]
[image]

MikeBrinson, 2015-04-22:
Spice4qucs documentation.

I have uploaded an outline of the spice4qucs documentation structure to the Qucs-Help repo. - 
see https://github.com/Qucs/qucs-help.git. -- branch spice4qucs. The style is identical to the new Qucs_Help documentation.

An example of the front page, in HTML, is shown below:
[image]

guitorri, 2015-04-22:
@MikeBrinson the building of the documentation branch is now enabled as a preview: http://qucs-help.readthedocs.org/en/spice4qucs/index.html 

For some reason the outline is a bit messed up. Perhaps because you added also the temp (.rst~) files to the repository?

ra3xdh, 2015-04-25:
Added Doxygen documentation entries for new classes and namespaces. See my last commits on `spice4qucs` branch.

guitorri, 2015-04-25:
@ra3xdh you might want to use the new [license header](https://github.com/Qucs/qucs/wiki/Development#license-header). It makes it easier/shorter to add multiple authors, one per line... Anyway, this should be updated everywhere in the code some day...

ra3xdh, 2015-04-26:
Added Xyce Parallel (with openMPI) support:
[image]
[image]

ra3xdh, 2015-04-29:
Added first part of examples. See my last commits. Examples are in `examples/ngspice` and `examples/xyce` subdirectories. Every example has comment on schematic.
The list of examples:
1. B-src.sch --- B-type sources usage. Voltage multiplier.
2. BJT-swp.sch --- Parameter sweep example
3. BJT.sch --- BJT 1-stage amplifier with example of euqations. (AC and TRAN analysis)
4. BJT_dc.sch DC sweep example. IV-chart of BJT.
5. EDD-Q-test.sch EDD with charge equations (nonlinear capacitance example)
6. RCL.sch Simple RCL circuit (AC and TRAN).
7. Realy.sch --- switching circuit with relay and inductive load.
8. Tunnel.sch --- Tunnel diode example. DC sweep usage with EDD.
9. audio_amp.sch --- Audio amplifier. Realistic circuit example.
10. irfz44_switch.sch --- MOSFET switching circuit.
11.  par_sweep_test.sch --- Parameter sweep (AC+TRAN) with RCL circuit
12. quarz.sch --- Crystal resonator subcircuit
13. quarz_test.sch --- Test of subcircuits with spice4qucs.
14. LM358_spice_prj.tar.gz --- A project example with Spice-netlists. 
15. hb_test.sch --- Harmonic balance simulation with Xyce

MikeBrinson, 2015-04-29:
Hello Vadim
I have added the XSPICE Cmeter component to spice4qucs.  This works byconnecting its input node to a circuit node. If one or more standard capacitors are attached to 
the circuit node the Cmeter outputs their total capacitance value.  The attachment gives an example of how it works.
As it is a measurement probe I have added it to the Qucs probe group with the icon colour codedto indicate it only works with ngspice. 

I will also push the RCLadder example to examples/ngspice with an added comment.

One other point - I have noted that the netlist button on the spice simulation window has a typo. Mike Brinson

mbrin72043@yahoo.co.uk

ra3xdh, 2015-04-30:
@MikeBrinson , fixed typo and added missing archived example project.

MikeBrinson, 2015-04-30:
Hello VadimThanks for the changes.
I have done some more testing of spice4qucs and find that it is broken.The problem seems to be the result of the change on the 26 April calledEnable double quotes in properties .....  Spice4qucs downloaded versions 
after this change appear not to work.  Component property texts are all mixed up -the attached examples show what happens.  This results in ngspice simulation failure.In some cases, using standard components and the custom ngspice simulation, simulationworks.  However, on plotting any data spice4qucs core dumps and crashes.
I have tried the latest spice4qucs code on both 32 and 64 bit Lubuntu Linux and the same 
result happens using either implementation.
Would you please check to see if you can reproduce this bug.
Best wishes.
Mike
 Mike Brinson

mbrin72043@yahoo.co.uk

ra3xdh, 2015-04-30:
@MikeBrinson, Crashes on opening diagram take place only if there are no datasets in current directory. This should be fixed in my last commit on `spice4qucs`. Also there was three schematic with wrong default dataset (specified inside XML *.sch --- this take place when schematic was renamed with File Manager instead of using "SaveAs" function ). This also was reason of crashes. These schematics are repaired. Now crashes should disappear for any combinations of datasets and broken schematics. The reason of this bug was in this commit https://github.com/Qucs/qucs/commit/8549c8bf5431ccdec009a8443027e178d5a949f3

I cannot reproduce bug when opening schematic with Custom simulation. All my circuits with custom simulation can be opened correctly. Please provide me sample of such circuit. I will  fix bug if confirmed or repair *.sch. All examples from example directory also are opened correctly. Your report about broken schematics in examples is very strange. There are no missing properties in components. Maybe at some point old schematic became incompatible. Send them to me and I will repair them.

ra3xdh, 2015-05-01:
I begin work on Chapter 2 "Basic Qucs, ngspice and Xyce simulation" of our documentation. I will push it in `spice4qucs` branch of `qucs-help` repository next week.

MikeBrinson, 2015-05-01:
Hello Vadim
Thanks for your good news on spice4qucs documentation preparation.
i have almost finished Chapter 1.  This outlines the background to the reasons forspice4qucs and the advantages it will bring to Qucs.  At the end of Chapter 1. I will 
include an example which shows how all the parts of the package fit together.  A fewadditional notes will (1)  introduce the tutorial nature of Chapter 2., and (2) introduce 
the scope of the reference material presented in later chapters.
I think we should also add a list of authors and the usual GPL copyright notice after the 
title on the first page.  Do you agree?
Best wishes.
Mike
 Mike Brinson

mbrin72043@yahoo.co.uk

ra3xdh, 2015-05-01:
>I think we should also add a list of authors and the usual GPL copyright notice after the title on the first page.  Do you agree?

Yes, I agree.  This information can be placed.

MikeBrinson, 2015-05-01:
Hello Vadim
Thanks for fixing the core dump bugs.I have downloaded the latest code and run it.All seems back to normal again.   If I find morebugs I will email you.
Best wishes.
Mike
 Mike Brinson

mbrin72043@yahoo.co.uk

MikeBrinson, 2015-05-01:
Hello Vadim
I have pushed a revised version of Chapter 1 of the spice4qucs-help document to the spice4qucs repo.This includes about 90% of Chapter 1 plus a revised index page presenting the author and copyright information.My next task is to build the introductory example outlining the new features introduced by the spice4qucs initiative.I will try and find time for this over the next few days.
Have a good weekend.
Best wishes.
Mike
 Mike Brinson

mbrin72043@yahoo.co.uk

ra3xdh, 2015-05-02:
I have pushed first part of Chapter 2 in `spice4uqcs` branch of qucshelp. Chapter 2 contains the following sections:
2.1 Introduction
2.2 Supported simulators
2.3 General simulation methods
2.4 Variable names
2.5 DC
2.6 AC
2.7 TRAN
2.8 Other forms of simulation

MikeBrinson, 2015-05-02:
Hello Vadim
Thanks for the Chapter 2. text and images.
They look great.I will add the Qucs yellow icon after the chapter heading and edit it so it follows the styleof Chapter 1.
We are making real progress towards launching version 0.0.19.
If I can find time over the next few days I will add more sections to thedocumentation, add more SPICE models to the package and undertake the first tests a series of scripts for ngspice Custom simulation.
Best wishes.
Mike Mike Brinson

mbrin72043@yahoo.co.uk

guitorri, 2015-05-04:
Hi @MikeBrinson and @ra3xdh, the documentation is looking good! 
Ii would be nice if you could add the schematics you used to generate the images in the documentation repository. Later on someone might want to either translate the documentation or rerun the examples...

MikeBrinson, 2015-05-04:
Hello GuilhermiAdding the documentation examples to the documentation repository is planned.At the moment we are at a very early stage in producing the documentation - giveus a few more weeks and the overall structure will become a lot clearer.  As we progressthe new documentation will provide both an interesting document for beginners and a valuablesource of reference information.
We will keep everyone informed of our progress.
See new pushes to the repo. over the next few days.
Best wishes.
Mike
 Mike Brinson

mbrin72043@yahoo.co.uk

ra3xdh, 2015-05-04:
@guitorri , Example of the test RCL circuit is in `examples/ngspice/RCL.sch` . I will add a link to this circuit in documentation.

ra3xdh, 2015-05-06:
Hello all,

I have added the following improvements in spice4qucs subsystem

* Fixed bugs in Libcomp, imporved implementation of BJT in qucs2spice
* Added experimental support of spice netlist embedded in libraries. You can find an example of such library and test this feature here: https://github.com/ra3xdh/qucs-rus-complib/blob/master/Opamp.lib This library contains embedded spice netlist for opamps within `<Spice>...</Spice>` section . There are two opamps in this example library. I don't find side-effects of this feature. 
* Implemented progressbars in simulation dialogs. 
[image]
* Added different sweep variable steps support for different transient simulations. This circuit illustrates how different steps are used. TRAN Par.sweep has 3 steps, but AC Par.sweep has 4 steps.
[image]
* Added DC sweep example with 1N4148 diode from Chapter 2.

ra3xdh, 2015-05-24:
Added `DISTORTION` and  `FOURIER` analysis examples:

`examples/ngspice/BJT-four.sch`
`examples/ngspice/Distortion.sch`

ra3xdh, 2015-05-24:
Added Chapter 4 of documentation. This chapter covers equations usage with spice4qucs for parametrization and  postprocessing purposes. All is available at `spice4qucs` branch of `qucs-help`.

ra3xdh, 2015-05-31:
New features and components addition is finished. At current state `spice4qucs` branch could be frozen. Only bugfixes will be added. We take course on rebase and merging in `master`. Documentation preparation will be continued until all new features will be documented.

ra3xdh, 2015-06-05:
I have tried to rebase spice4qucs today. Here is rebased  `spice4qucs` branch https://github.com/Qucs/qucs/tree/rebase_spice4qucs. Please test all again carefully. Don't merge it! For unknown reason automatic rebase of QTextEdit->QPlainTextEdit replacement went wrong. It was need to replace QTextEdit->QPlainTextEdit manually. Please check this commit https://github.com/ra3xdh/qucs/commit/f656a946b0679b8c2f04f7d4fed987d115538c95 especially.

ra3xdh, 2015-06-12:
`rebase_spice4qucs` proposed for merge. See #275

ra3xdh, 2015-07-22:
Updated tasklist according recent discussion .

guitorri, 2015-07-24:
Have you made any progress on testing?

ra3xdh, 2015-07-24:
@guitorri  I will try to add test this weekend. It's no sense to add Xyce in tests, because there is no such package in Linux distributions. Ngspice tests require `ngspice` installed. I think the following test projects will be sufficient for the first time:
1. RCL-circuit (AC+TRAN)
2. BJT amplifier (AC+TRAN+Equations)
3. BJT amplifier (AC+TRAN+Parameter sweep)

felix-salfelder, 2015-07-24:
> no such [xyce] package in Linux distributions

FWIW: there is a draft (here)[http://anonscm.debian.org/cgit/pkg-electronics/xyce.git/]. its not official (nobody ever cared) yet but it worked the last time i built it. yes, you need the trilinos package. it's also in the works, haven't touched that for a year or two now...

ra3xdh, 2017-06-29:
Closing, because #275 is closed. Use issue tracker at my home repository to report issues on Qucs-S.

closed 2017-06-29 by ra3xdh