Qucs/qucs #230: Random crashes when closing a Tab

in3otd, 2015-04-19:
I get "random" crashes, when closing a Tab. Did not yet find a repeatable sequence to reproduce the bug, but after opening/simulating/closing a few tab Qucs crashes.
Here is a gdb backtrace of a crash happened when closing a tab, don't know how valid it is

````
0  0x00000003 in ?? ()
#1  0x08071331 in QWidget::setHidden (this=0x91ecb30, hidden=true)
    at /usr/lib/qt/include/QtGui/qwidget.h:495
#2  0x0807b9ea in QucsApp::slotHideEdit (this=0x91dce20) at qucs.cpp:2630
#3  0x080839cd in QucsApp::closeFile (this=0x91dce20, index=1) at qucs.cpp:1623
#4  0x08083ea4 in QucsApp::slotFileClose (this=0x91dce20, index=1) at qucs.cpp:1613
#5  0x0810a99e in QucsApp::qt_static_metacall (_o=0x91dce20, _c=QMetaObject::InvokeMetaMethod, _id=12, 
    _a=0xbf958938) at qucs.moc.cpp:269
#6  0xb74eedd5 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) ()
   from /usr/lib/libQtCore.so.4
#7  0xb6eac665 in QTabWidget::tabCloseRequested(int) () from /usr/lib/libQtGui.so.4
#8  0xb6eac8d9 in ?? () from /usr/lib/libQtGui.so.4
#9  0xb74eedd5 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) ()
   from /usr/lib/libQtCore.so.4
#10 0xb6ea0325 in QTabBar::tabCloseRequested(int) () from /usr/lib/libQtGui.so.4
#11 0xb6ea03c9 in ?? () from /usr/lib/libQtGui.so.4
#12 0xb74eedd5 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) ()
   from /usr/lib/libQtCore.so.4
#13 0xb70d7f6d in QAbstractButton::clicked(bool) () from /usr/lib/libQtGui.so.4
#14 0xb6df0881 in ?? () from /usr/lib/libQtGui.so.4
#15 0xb6df1c17 in ?? () from /usr/lib/libQtGui.so.4
#16 0xb6df1d1e in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /usr/lib/libQtGui.so.4
#17 0xb6a372ca in QWidget::event(QEvent*) () from /usr/lib/libQtGui.so.4
#18 0xb6df2d22 in QAbstractButton::event(QEvent*) () from /usr/lib/libQtGui.so.4
....
````

yodalee, 2015-04-19:
The crash is happened when you focus on the textedit of component property, and try to close file. Directly call slotHideEdit() will cause crash though the reason is not clear.
Add one line of editText->clearFocus(); can solve this problem.
But there is more, in some place of qucs.cpp it still call editText->setHidden(true);, they will still crash. So we need to change all direct call to call slotHideEdit()

in3otd, 2015-04-29:
Fixed by #243

closed 2015-04-29 by in3otd