Qucs/qucs #233: Slow graph drawing of long simulation runs

nvdl, 2015-04-20:
The graph view brings schematics to a freeze if there are a lot of data points to plot.
Can it be configured to just plot enough points from the data set as the number of visible pixels on the screen? In this way long simulation results will be plotted quickly and schematics update will be faster if scrolled.

guitorri, 2015-04-21:
No, there is no code to optimize points to the visible pixels... The slowdown is related to legacy code used on the graphical rendering. All graphic items (including points) are painted/updated in a loop, too many items and the loops freezes everything. It will take a while to replace the legacy code and fix this issue...

In the mean time, there are a few things you can try to speedup the rendering:
 - Try reducing the number of points (simulation steps) to the minimum acceptable. 
 - You can try to put one graph per data display files (.dpl), which open in separated tabs, to keep the schematic more responsive. But if you have many datapoints (>10000 or so) it will be slow when scrolled. 
 - You can try to use an equation block to re-sample the result of interest. Something like `y=interpolate(signal,time,100)` should give you an interpolation of the `signal(time)` vector with 100 samples in the vector `y`. Please test it first, I am not sure if this will work...

nvdl, 2015-04-22:
Is there a possibility to plot the results externally?
I have not dug into the simulation file format or do not know if this is the place to ask about it.

Contrary to other (but not all simulators), the simulation needs to be run again if a new net label is added. Does it imply that simulation run only saves the labelled nets' results?

If some help is needed in graphing, I can get started with it.
But need to be pointed in the proper direction in the beginning.

guitorri, 2015-04-22:
There are MATLAB/Octave and Python scripts. 

For Octave: 
- You can use the Octave console from Qucs if you like. It can also call a script after the simulation.
- See https://github.com/Qucs/qucs/tree/master/qucs/qucs/octave
- Here is another example: https://sourceforge.net/p/qucs/mailman/message/27347746/

To load the data with Python:
- https://github.com/Qucs/qucs/tree/master/qucs/qucs/python
- https://github.com/Qucs/qucs-test/blob/master/qucstest/qucsdata.py

If I recall correctly only sources, probes and labeled nodes are saved by default.

The mailing lists and forums are perhaps better suited for general questions: https://sourceforge.net/p/qucs/mailman/

guitorri, 2015-04-22:
Forgot to add that you can also use the command line tool `qucsconv` to convert the Qucs datafile to MATLAB (.mat) format and CSV files

nvdl, 2015-04-26:
**qucsconv -i in.dat -o out.csv -if qucsdata -of csv -d net1.Vt**

The CSV file is colon ";" separated rather than comma.
Any good reason for that?

in3otd, 2015-04-26:
...never noticed that before, I think this is due to the country of origin of the first developers of `Qucs`, where the decimal separator is a comma and so a semicolon is used as field separator in `.csv` files.
But I see that `qucsconv` uses anyway a full stop as decimal separator; don't know how good/bad this combination works in practice.
I see in the source code that the function writing the `csv` format can accept a different separator, but in the main program it is hard-coded as semicolon. Maybe we could add an option or even try to detect the user locale.

nvdl, 2015-04-26:
**CSV file is semicolon delimited**

I agree. I deleted the comment though. :D

It was about just being specific what a CSV stands for.
We can keep it as a hint here so that it is user configurable later.

guitorri, 2015-04-27:
In some places it is called "character separated values". The delimiter can be anything (space, tab,...), as long as it do not mess with the data...

closed 2018-07-19 by guitorri