Qucs/qucs #234: Flipped Y range in Cartesian diagram

nvdl, 2015-04-20:
Cartesian diagram accepts flipped range for the manual limits of left axis.
For example, start = -6 and stop = -9 does not give any warning.

in3otd, 2015-04-21:
...well, I think that actually this was intended as a feature, so you can invert the direction of the axes, if needed. The alternative would have been to force start < stop and add a checkbox 'invert axis'...

nvdl, 2015-04-22:
Inversion of axis should flip the plot as well but that does not happen.

Another concern is the step that can be -ve or +ve.
I think +ve (absolute) step should be allowed only as that is enough to let the grapher know.

in3otd, 2015-04-22:
with the cartesian diagram the plot is indeed flipped when inverting the y axis here. With which one does it not work there?

The rationale behind the automatic step sign determination is to avoid bothering the user with details like this, that can often be entered wrongly and for which the actual intended behavior is obvious. I mean, I do not find useful to show an error message telling "step sign is wrong!", having him pressing "Ok", entering the right sign and finally obtaining the result he wanted.
If we force the step to be entered as absolute value, when a mathematically-oriented user correctly enters a negative step when needed, again, it will make no sense to show an error message telling "enter the step as absolute value". Qucs figures out the step sign by itself and avoids bothering the user.

nvdl, 2015-04-22:
I agree with the step value being +ve or -ve.

For the axis inversion, it works as I tested now.
But, before closing it, I want to double check one simulation to which I do not have access now.

closed 2015-04-26 by nvdl