Qucs/qucs #238: Component rotate after drag start disappears

nvdl, 2015-04-22:
A component disappears after the dragging is started and Ctrl+R (to rotate) is pressed.

Step to reproduce:
Select a component. Start dragging it using left mouse button (pressed).
Press Ctrl+R.

Version: 0.0.19
Branch: Master

in3otd, 2015-04-22:
yes, this was already in 0.0.18, unfortunately. (BTW, you can rotate a component while placing it using the mouse right button)
It is likely part of the small and big annoyances brought by the Qt4 porting process, where the components painting flow had to be significantly changed.
I think you will find other things like this so I'll suggest to keep all the schematic drawing issues together, you can update this issue when you find something else not working as expected

I can add another one already 
- [ ] While moving a Text component, the boundary rectangle is always drawn at the origin; this makes very difficult to move it to the desired position

nvdl, 2015-04-22:
I cannot follow "so I'll suggest to keep all the schematic drawing issues" properly.
Is there any other thread where I should post these issues?
I am trying to bring them up so that we may not skip them when polishing the interface.

It has already been mentioned in #145 but not explicitly as an issue.
I am new to GitHub, so please let me know if creating new issue for every minor thing is not a good practice.

This issue also exists in 0.0.19; the current master branch.

in3otd, 2015-04-22:
no problem, I am also relatively new to GitHub. And every project has its own way to handle issues/pull requests/development, etc. 
Since I expect that you will find more things not working properly in the graphics handling and all will probably more or less related, I was suggesting to keep a single issue and update it with new things as you find them. Just edit your first post and add an item to a list.

nvdl, 2015-04-22:
**Ghosting when placing or dragging a component**

[image]

Version: 0.0.19

nvdl, 2015-04-23:
**Lost schematic**

The schematic is lost after these steps:

Right click in an empty place in the schematic -> Edit Circuit Symbol

in3otd, 2015-04-23:
I cannot reproduce this here (w/ Linux); which OS are you using?
What do you mean by 'lost'? The application crashes? If not, If you press F9 again can't you see again the schematic?

nvdl, 2015-04-23:
Wrongly reported. It switches to paintings. Pressing F9 brought it back to schematics.

nvdl, 2015-04-23:
**Cartesian diagram edit dialog**

The dialog form to edit the properties of cartesian diagram cannot be moved to show the diagram if it is above it. Instead, the whole application form moves.

in3otd, 2015-04-23:
I cannot reproduce this here (w/ Linux); which OS are you using?
I can freely move "Edit Diagram Properties" of a cartesian diagram as an independent window.

nvdl, 2015-04-23:
I see it in 0.0.18 and 0.0.19.

Double click on the cartesian diagram and try to drag it. The main form changes from maximized state to normal state and starts dragging along keeping the dialog form at the same position.

OS: Debian, gnome-shell.

Does it has something to do with other GUI libraries?

in3otd, 2015-04-23:
so, just to be sure, you have the main Qucs window maximized, the current Tab is a Data Display with a cartesian diagram, you double click on it, the "Edit Diagram Properties" dialog opens and when you try to move it to see the underlying cartesian diagram the main window is no longer maximized and is dragged instead of the "Edit Diagram Properties" dialog ?? With Slackware and xfce here, it behaves normally, the "Edit Diagram Properties" dialog can be moved and the underlying main windows stays maximized. I know that the Qt behavior is (usually slightly) different on different platforms and even differs between different window managers on Linux. I do not have Debian installed, but might be able to find a Ubuntu machine with Gnome to try. Or, I can try to run Debian in a VM, but it will take some time to set it up.

nvdl, 2015-04-23:
[image]
[image]

It can **also** be recreated by starting afresh and placing the diagram. The diagram properties dialog shows up. If you try to move it, the pointer jumps to the title position of the main window and both start moving together.

in3otd, 2015-04-23:
...interesting behavior... Those are modal dialogs and, as you may already suspect, this may be a Gnome "feature". Please take a [look at this] (https://www.youtube.com/watch?v=qZ2x0RmlQlE) and check if it changes the windows behavior to a more "normal" way.

All those windows do not remember their size and I agree it might be useful sometimes; there is PR currently open (#232) which changes a bit the way the data are shown in the tables, to avoid to have to re-size too often.

nvdl, 2015-04-23:
Yes, that was GNOME issue. I had a chuckle at how the guy in the video was complaining.

Here is the fix:

**gsettings set org.gnome.shell.overrides attach-modal-dialogs false**

All is well now.

nvdl, 2015-04-23:
**Zooming is not stable**

Ctrl+MouseWheel zooms properly as long as the wheel is moved slowly.

Fast wheel movements make the schematic zoom in and out even though the wheel movement is in one direction.

nvdl, 2015-04-23:
**Drawing nets**

Now, the nets are drawn as:

[image]

It will be more user friendly if the nets are drawn as:

[image]

The issue with the current way of drawing is that one has to click on three points rather than two points to route a net in the generally more desired way.

guitorri, 2015-04-24:
After you start the wire, right-click toggles the orientation. A key shortcut might be a good idea.

nvdl, 2015-04-24:
Yes, the right click works. I will try to add the space key shortcut.
Do we have shortcuts management? It will be nice if I get pointed to its source files.

How does the collaboration work? Should I create my own branch or fork it?

I edited the routing cross size so that it does not go through the whole schematic but plan to make it user configurable.

guitorri, 2015-04-24:
There was a shortcut manager in development by @yodalee, I am not sure what is the status. This is the [branch](https://github.com/yodalee/qucs/tree/feature54-custom-shortcut) I guess.

It is best if you use your fork, and send us pull requests. At least in the beginning till we get to know you :smile:

nvdl, 2015-04-24:
Being new to Git, how does the merge go?

I need his work to get the shortcut or I hardcode it.
If I hardcode it, should it be changed later to support the shortcut manager?

yodalee, 2015-04-25:
You can use the shortcut manager I wrote, just fetch the branch and apply to your repository. But be careful, I cannot guarantee that it's bug free.

nvdl, 2015-04-26:
I need a bit of help regarding this.
I have already forked the master branch.
How can I fetch and apply your branch to the fork?

Or should I fork your branch?

Branch name?

nvdl, 2015-04-26:
** Net disappears when dragged **

The net wire disappears when dragged to a new position making it "not so user friendly" to know where it will land.

in3otd, 2015-04-26:
I am not an expert, but I will say that jut to try the shortcut manager of Yodalee, you can do
`git clone -b feature54-custom-shortcut https://github.com/yodalee/qucs.git`

Or if you would plan to modify it, create your own branch and pull his custom shortcut branch:
`git checkout -b <a_new_branch_with_a_name_you_like>`
`git pull https://github.com/yodalee/qucs.git feature54-custom-shortcut`
assuming you were on the `Qucs` master branch you will need to resolve a merge conflict (hint: IMHO the conflicting parts in `qucs.cpp` should read

````
    saveApplSettings();
    clearShortcutMap();
````
)

finish the merge with `git commit`

Then compile `Qucs`:
- `./autogen.sh`
- `./configure --enable-maintainer-mode --prefix=<path_where_you_would_like_to_install_this_Qucs>`

run `qucs` from the install directory

Select `File` -> `Application Settings` -> `Custom Shortcut`
...have fun (never played with this branch before, so cannot help much...)

nvdl, 2015-04-27:
Thanks for the info. I am changing the icons and updated the ".png" files in:

**qucs/qucs/qucs/bitmaps**

After recompilation, they are not updated. Is there a resource file that is created out of them.
Another things is that some of the icons are at multiple locations e.g.

**qucs/qucs-help/bitmaps**

also has some icons.

Has anyone looked into it?

nvdl, 2015-04-27:
**Drop down list insensitivity**

[image]

The click does not work at the location in the picture.
Possibly it is due to the label "y-Axis:" protruding to its region and stealing the mouse events.

in3otd, 2015-04-27:
uhm, might this be another Gnome "feature" ? :grin:
here clicking on the drop down list text works fine; is the `Style:` drop down list on the same diagram behaving the same? Can you try with another window manager?

nvdl, 2015-04-27:
I am not seeing this behavior with "Color:" and "Style:" boxes.
So I ruled out the GUI "feature".

It works fine but not on the left-ish side. Note the pointer position.

Previously, I was considering it as a sluggish but there is a region where click does not work at all.

nvdl, 2015-04-27:
**Open with qucs does not work (OS integration)**

If a schematic name is passed as the first argument, qucs GUI does not start due to missing command line parameters.

It should start GUI if the file exists and is a valid schematic.

in3otd, 2015-04-28:
Regarding *Open with qucs does not work (OS integration)* you have to use the `-i` option, e.g. `qucs -i mydesign.sch`. You can see all the options using `qucs -h`. We should likely print the usage help in case an option is not understood, though.

in3otd, 2015-04-28:
Regarding the *Drop down list insensitivity* I have installed Debian 8.0.0 with Gnome in a VM and I cannot see that behavior, the drop-down list opens also when I click in the middle (or at the beggining) of the text.
But a few days ago some (small) changes to that part of the dialog were made; these should not affect the drop-down list behavior but you might want to update to the current code, if not already done, and re-check.
I guess anyway your system configuration (window manager ?) is not the same as I had on the Debian I installed: I see that the `Color:` button is not colored (known issue on some WM) but it was on the version I installed...

nvdl, 2015-04-28:
I checked and the issue is gone in 0.0.19 and color button is also colored accordingly.
The issue exists in 0.0.18.

nvdl, 2015-04-28:
**Getting rid of the outdated icons**

[image]

Work in progress.

nvdl, 2015-04-28:
**Slow undo redo**

Undo, redo is slow even if a single component is moved.

nvdl, 2015-04-28:
**Symbols in font**

Is there any font requirement for the degree symbol?

[image]

nvdl, 2015-04-29:
**Rotate and mirror arrows scale with zoom level**

Looks like the zoom level is not considered when they are drawn.
Should be replaced by icons.

guitorri, 2015-04-29:
@nvdl at some point we will have o split this big issue into individual issues... it is becoming hard to read and reply to it.

There is a CLI command in qucs to output the icons for the components, I am not sure we should change them at the moment.

For the icons, I hope you are using some opensource artwork. And yes, all the images are embedded in the binary via the .qrc files.

Qucs-help is probably going to be deprecated in favor documents generated wit a new system https://github.com/Qucs/qucs-help

nvdl, 2015-04-29:
@guitorri:

It was suggested by @in3otd to keep one issue and accumulate. That is why I am not issue spamming.
It is getting tricky for me to keep track of this issue as well now.

Should I go for new issue next time?

**All icons are either GPL, free for non-commercial or free for commercial**

in3otd, 2015-04-29:
yes, I was suggesting to group the confirmed issues in the first post: just edit it, add a description of the issue, maybe with a check-box, so we can tick it when and if it will be solved.

Regarding the "outdated icons" I think @nvdl was referring to the toolbar icons. Since "outdated" is an opinion, I think this should be discussed more widely before starting to making/proposing changes. The fact that these icons have been the same for 10 years it's a good argument to keep them, not to change them, IMO.
What could be done is to adapt the icon to the local theme in use (but for Windows we need to provide our own icons), see http://doc.qt.io/qt-4.8/qicon.html#fromTheme .

Which icons are needed in the menus? Currently only the menu items which have a corresponding button in the toolbar have an icon, the same as in the toolbar (the same QAction is linked).

nvdl, 2015-04-29:
I am thinking of loadable icon pack support so people can keep what they like.
For me these are really old and very difficult to bear.

nvdl, 2015-04-29:
**Component disappears if rotated after dragging starts**

**Moved from the top.**

A component disappears after the dragging is started and Ctrl+R (to rotate) is pressed.

Step to reproduce:
Select a component. Start dragging it using left mouse button (pressed).
Press Ctrl+R.

Version: 0.0.19
Branch: master

nvdl, 2015-04-29:
**Intelligent net orientation during drawing**

The net orientation should be in a way follow the mouse.

if ( abs(x-of-last-point - x-of-current-point) > abs(y-of-last-point - y-of-current-point) )
  drawHorizontallyAndThenVertically()
else
  drawVerticallyAndThenHorizontally()

nvdl, 2015-04-29:
**RMB panning support (shortcuts manager)**

Need to add this for convenient (as I think) panning.

in3otd, 2015-04-29:
*Intelligent net orientation during drawing*: yes, other SWs use this strategy to draw wires, I personally find it more intuitive than the current one in Qucs, so it will be  good addition, IMHO.

*RMB panning support*: pan support with mouse will be nice also, I'm used to use the CMB for this, RMB is already used for the context menu. Then we will have to decide if we move the schematic or the viewport while panning. (i.e. you move the mouse up, will you show more of the upper part of the schematic or will you move the schematic, so that its bottom portion is now shown?)

In general I am quite cautious about GUI changes, since I assume there is already a wide user base used to a certain GUI appearance and behavior, so I would like to hear more opinions about this.

nvdl, 2015-04-29:
__i.e. you move the mouse up, will you show more of the upper part of the schematic or will you move the schematic, so that its bottom portion is now shown?__

Can you explain it more?

nvdl, 2015-04-29:
**Ghosting when rotating a component**

Repaint is missing after RMB is used to rotate the "to be placed" component.

nvdl, 2015-04-29:
**Text disappears after drag starts**

nvdl, 2015-04-29:
**Text rotation is not around center**

Text moves like a snake when rotated.

nvdl, 2015-04-29:
_Which icons are needed in the menus? Currently only the menu items which have a corresponding button in the toolbar have an icon, the same as in the toolbar (the same QAction is linked)._

I do not see the icon in the "File" menu.

nvdl, 2015-04-29:
**Fixed offset of the component parameters' text**

[image]

To make the schematic look cleaner, the parameters' text needs to be movable.

nvdl, 2015-04-29:
**- [ ] "Zoom in" button should zoom in, need another button for region zoom**

Zoom in should be functionally inverse of zoom out. Region zoom should be another button.

in3otd, 2015-04-29:
*Symbols in font*: also here (linux and Windows), it has maybe something to to with how the strings encoding is handled by Qt (UTF-8 or Latin1 or...)

*icons in the "File" menu*: now I see what you mean. I see the icons in Windows, Debian+Gnome and Ubuntu+Unity but not in Slackware+Xfce. I guess it is another thing which is WM dependent. May I suggest you to install another Linux distribution in parallel so also you can check if the behavior differs in different distributions?

*RMB panning support:* if you put on RMB there is risk of panning while wanting to open the context menu. CMB press is currently free and already related to zoom/pan via Ctrl/Alt + scroll wheel.. Shortcut manager could potentially satisfy all tastes, of course.
Again why would you like to change when the context/menu appears? It currently appears after RMB press, what's wrong with that? Which problem are you trying to solve?
The "move the schematic or the viewport" was a half joke: the abstraction is that you are looking to a large schematic thru a small window (viewport). Some SW does panning by moving the window over the schematic and others by keeping the window fixed and moving the underlying sheet. It is the same, only the moving directions are swapped.

*Ghosting when rotating a component*: yes, also here.

*Text disappears after drag starts*: yes, it is the same issue as in my first reply at the top

*Text rotation is not around center*: yes, applies to every component whose bounding box is not symmetrical around its center. It was also mentioned in #200 

*Fixed offset of the component parameters' text*:  is `Move Component Text` (`Ctrl+K`) what you had in mind?

*Zoom in" button should zoom in, need another button for region zoom*: so you do not like that a simple click in the schematic does a zoom in and drawing a region does a region zoom. Ok, but...

*I think you should separate what are real issues (zoom not stable, ghosting when rotating, etc.) from your personal preferences (RMB action, icons, etc.).* As said, I am against imposing arbitrary changes to existing features to the silent majority, without hearing other opinions.

nvdl, 2015-04-30:
_It currently appears after RMB press, what's wrong with that? Which problem are you trying to solve?_
To allow panning functionality and the menu at the same time.

_no icons in menus in Debian+Gnome here, bad memory._
Same here.

_Text disappears after drag starts_
Fixed.

_Fixed offset of the component parameters' text: is Move Component Text (Ctrl+K) what you had in mind?_
My bad!

_Zoom in" button should zoom in, need another button for region zoom: so you do not like that a simple click in the schematic does a zoom in and drawing a region does a region zoom. Ok, but..._
Agree.

_I think you should separate what are real issues (zoom not stable, ghosting when rotating, etc.) from your personal preferences_
I am figuring out how to separate personal wishes from gloabal ones.

nvdl, 2015-05-04:
- [ ] **Menu icons**

For Gnome:

**gconftool-2 --type boolean --set /desktop/gnome/interface/menus_have_icons true**
**gconftool-2 --type boolean --set /desktop/gnome/interface/buttons_have_icons true**

nvdl, 2015-06-03:
**Proper component text placement after rotation or flipping**

The component text should be placed properly and not on top or too close to the component after rotation or flipping.

in3otd, 2015-06-03:
@nvdl do you have an example of component text not placed properly after rotation or flipping? I spent some time a while ago to fix this issue and I thought the current git code was working fine now...

nvdl, 2015-06-03:
@in3otd You mean the current master branch?

in3otd, 2015-06-04:
Yes, I mean the current master branch. I assume that issues logged on GitHub are always relative to the current master branch, or just a few commits before current. Anyway, the text rotation/flipping corrections were made a few months ago, IIRC; I was wondering if there is still some case not working properly.

nvdl, 2015-06-09:
[image]

Maybe I am too picky but it should look like the the ones in the 2nd row.

nvdl, 2015-06-10:
- [ ] **Upgrade old dialog box for save as image option**

Now, the dialog box that asks about the location for saving the image seems outdated (possibly Qt3 stuff). It should be upgraded to have a native OS look.

[14 more replies]

closed 2018-07-19 by guitorri