Qucs/qucs #240: Document somewhere the environment variables.

guitorri, 2015-04-23:
These are used to relocate or override the executables.

Some are setup during the Windows install.

Document them somewhere accessible (wiki, `qucs` CLI, ...)

These are the ones we have:
```
./main.cpp:647:  char* var = getenv("QUCSDIR");
./main.cpp:677:  var = getenv("QUCSATOR");
./main.cpp:685:  var = getenv("ADMSXMLBINDIR");
./main.cpp:702:  var = getenv("ASCOBINDIR");
./main.cpp:719:  var = getenv("OCTAVEBINDIR");
```

guitorri, 2015-04-24:
Related to #204

guitorri, 2015-09-17:
I guess we could add this to the `qucs -h` documentation...

felix-salfelder, 2015-09-17:
before documenting (yes, i agree on `qucs -h`, let me suggest to remove the unnecessary ones.

`OCTAVEBINDIR` could be obtained directly via `octave-config -p BINDIR`, `ADMSXMLBINDIR` and `ASCOBINDIR` look redundant to me. we really should not prominently encourage users to override or hardwire these. are there use cases?

so now, there's only QUCSDIR, and QUCSATOR, and as i understand QUCSCONV. these even start with `QUCS` (surprise! -- try `python --help`).

guitorri, 2015-09-19:
The variables are used on the [Windows installer](https://github.com/Qucs/qucs/blob/51e4d23548e7d6c2f9342c98b2070b7150a3419c/qucs/contrib/innosetup/qucs.iss#L64-L67).

Anyway, if the user can set external tools on the settings, why not also via environment variables?

felix-salfelder, 2015-09-19:
> The variables are used on the Windows installer.

this is really bad. lets deprecate them, not document any further.

> why not also via environment variables?

this is fine, but the names and functionality is misleading. a variable called `QUCS_ADMSXML` could be used to override the `admsXml` exec. currently it starts with `ADMSXML.*`, which sounds a lot like an adms option (which it is not) and it sets a path, although this is not required. same with the other 2.

guitorri, 2016-01-28:
I agree we should favor the use of PATH to find out the external tools. 
Let us sort this out on the next release cycle.