Qucs/qucs #244: parse_result.py: ComplexWarning: Casting complex values to real discards the imaginary part

nvdl, 2015-04-26:
[parse_result.py](https://github.com/Qucs/qucs/blob/master/qucs/qucs/python/parse_result.py)

Trying to plot transient analysis.

import parse_result as pr
data = pr.parse_file(datFName)
y = data[t20]

Gives the warning:
ComplexWarning: Casting complex values to real discards the imaginary part

t20.Vt
<type 'numpy.ndarray'>
[ 9.97959075+0.j  9.98024664+0.j  9.98082833+0.j ...,  9.98381485+0.j
  9.98381485+0.j  9.98381486+0.j]

Looking into the data file from simulator, there is no imaginary part but the returned numpy array has an imaginary part.

guitorri, 2015-04-27:
Have you tried this: https://github.com/Qucs/qucs-test/blob/master/qucstest/qucsdata.py
The parser you are using always pre-allocates storage as complex...

nvdl, 2015-05-03:
Yes, it works as expected.

closed 2015-05-03 by nvdl