Qucs/qucs #246: Opening a subcircuit proposes its symbol for pasting

in3otd, 2015-05-01:
To reproduce the issue:
- create a new project
- create a subcircuit in the project (just insert a couple of ports), save it.
- close the subcircuit schematic Tab
- open the subcircuit by double-clicking on its name in the project view
- the subcircuit schematic will open and **its symbol will be proposed for pasting, sticking to the mouse pointer**. This does not make sense, as one never needs to use the subcircuit symbol in its schematic.

@yodalee can you take a look at this? This behavior appeared after merging #197, before that, opening a subcircuit did not automatically propose its symbol for pasting.

yodalee, 2015-05-01:
I cannot reproduce this. When I double click on its symbol, it open the `edit component properties` dialog.

yodalee, 2015-05-01:
Oh, I got it. This issue is more complex than I thought.
Here is the story: in #197 I replace the QTreeWidget with custom QTreeView, and I think that, QTreeView handle double click event less smart then QTreeWidget, which will also fire the single click event. 
Here is some discussion about this, I will try to fix this on weekend:
http://www.qtcentre.org/threads/7858-Double-Click-Capturing
http://stackoverflow.com/questions/4627347/distinguish-between-single-and-double-click-events-in-qt
I think Qucs may face similar problem when switching to `QGraphicsView`

ra3xdh, 2015-06-03:
There may be another way to fix this issue. We can simple forbid pasting subcircuits from TreeView into itself! You can look at my concept in my test branch: https://github.com/ra3xdh/qucs/tree/246_fix I have tested it and this bug seems to be fixed. Please test it to find undiscovered side-effects.

ra3xdh, 2015-06-04:
@in3otd Thank you for testing! I have just prepared PR #269 for this fix.

in3otd, 2015-06-20:
Fixed in #269

closed 2015-06-20 by in3otd