Qucs/qucs #247: "Save all" does not remove the "diskette" icon from all tabs

in3otd, 2015-05-05:
When more tabs are open and modified (so they show the small "diskette" icon near the file name in the Tab), doing a "Save all" remove the diskette icon only from the current Tab, not from the other ones. All the modified files are correctly saved, though.
I think this appeared after #197 .

yodalee, 2015-05-07:
Actually I think this is appeared after #195 
There are many way to fix it, but require discussion.
The cause is that `QucsApp::slotFileChanged` only update current tab status. Function `saveAll` will call each tab save their file w/o change to them. So the status update will only update to current tab.
We can let `saveAll` call `DocumentTab->setCurrentPage()` every time, but that might cause some flash during execution. The other way is make `slotFileChanged` accept the document as a parameter, so it can distinguish which tab should be updated. But this deepen the connection between `QucsApp` and `QucsDoc`

in3otd, 2015-05-09:
I think that `slotFileChanged` needs to know which document has sent the signal; currently when you do a *Save All* every document emits the `signalFileChanged` but `slotFileChanged` updates only the current Tab every time, regardless of which document emitted the signal - this is surely not a good thing.

Or maybe, since the *Save All* is the only case when the diskette icon of non-current tabs is modified (right?) `QucsApp::slotFileSaveAll()` could call `DocumentTab->setTabIcon` for every document that returned true for `saveFile(Doc)`, but seems less clean to me.

closed 2015-09-19 by guitorri