Qucs/qucs #248: HB simulation regression

guitorri, 2015-05-05:
@MikeBrinson reported this some time ago. I forgot to create the issue.

At some point the HB simulation started to behave differently from version 0.0.18.

I added the schematic to the testsuite but could not investigate it further, see branch 'add-HB-test':
https://github.com/Qucs/qucs-test/tree/add-HB-test

Doing a git bisect the regression appears around:
git bisect good d66af600faf7bb4733b3999a3e3c42666f7a9a89
# first bad commit: [eb313816e5f3d35e48ab855b3bfa7ecca7f93374] Fix a crash

guitorri, 2015-05-30:
@MikeBrinson, the regression on the HB simulation should be fixed now.
Have you managed to compare (the limited) qucsator HB w.r.t ngspice or xyce?

closed 2015-05-30 by guitorri

MikeBrinson, 2015-05-31:
Hello Guilherme
Thanks for finding the bug in Qucs HB.Now that I have finished work on spice4qucs for the next releaseI will run tests on Qucs HB and Xyce HB.  Earlier tests (using Qucs-0.0.18)and Xyce showed there was general agreement between the simulators.However, we need to do much more in the way of investigative tests to confirmmy earlier results.
Vadim and I have written a paper on this subject which has been acceptedfor MIXDES 2015. This will be presented at the end of June in Poland.
I will start testing your latest bug fix this week.As soon as I have some results I will email them to you.
Best wishes.
Mike
 Mike Brinson

mbrin72043@yahoo.co.uk

   
Closed #248 via fde73e5.—

commit 51e4d235 (2015-05-30): Merge pull request #257 from guitorri/248-HB-regression

Resolve #248 - HB regression
commit fde73e5f (2015-05-30): Resolve #248 - HB regression

The issue was introduced in 9bd368d1.
The 'r++' increment was left behind during the refactor.