Qucs/qucs #249: Non-const reference cannot be defaulted in the class body

nvdl, 2015-05-06:
**ADMS installation:**
http://sourceforge.net/projects/mot-adms/files/adms-source/2.3/adms-2.3.4.tar.gz/download
extract
sh bootstrap.sh
./configure --enable-maintainer-mode
sudo make install

**Qucs core:**
cd qucs-core
./bootstrap.sh
./configure --enable-maintainer-mode --with-mkadms=/usr/local/bin/admsXml
make clean
make

**Error in compilation:**
```
Making all in interface
make[3]: Entering directory `/home/nvd/software/simulators/qucs/github/qucs/qucs-core/src/interface'
/bin/bash ../../libtool --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I../..  -I../../src -I../../src/math -I../../src/components -I../../src/components   -g -O2 -std=gnu++11 -O0 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -W -Wall -MT qucs_interface.lo -MD -MP -MF .deps/qucs_interface.Tpo -c -o qucs_interface.lo qucs_interface.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I../../src -I../../src/math -I../../src/components -I../../src/components -g -O2 -std=gnu++11 -O0 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -W -Wall -MT qucs_interface.lo -MD -MP -MF .deps/qucs_interface.Tpo -c qucs_interface.cpp  -fPIC -DPIC -o .libs/qucs_interface.o
In file included from ../../src/nasolver.cpp:47:0,
                 from ../../src/nasolver.h:150,
                 from ../../src/trsolver.h:28,
                 from e_trsolver.h:38,
                 from qucs_interface.cpp:49:
../../src/nodelist.h:47:3: error: 'qucs::nodelist_t::nodelist_t(qucs::nodelist_t&)' declared to take non-const reference cannot be defaulted in the class body
make[3]: *** [qucs_interface.lo] Error 1
make[3]: Leaving directory `/home/nvd/software/simulators/qucs/github/qucs/qucs-core/src/interface'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/home/nvd/software/simulators/qucs/github/qucs/qucs-core/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/nvd/software/simulators/qucs/github/qucs/qucs-core'
make: *** [all] Error 2
```

guitorri, 2015-05-06:
Which compiler version?

nvdl, 2015-05-06:
**G++**:

Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.7/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 4.7.2-5' --with-bugurl=file:///usr/share/doc/gcc-4.7/README.Bugs --enable-languages=c,c++,go,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.7 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.7 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --with-arch-32=i586 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.7.2 (Debian 4.7.2-5) 

**GCC (if needed)**:
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.7/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 4.7.2-5' --with-bugurl=file:///usr/share/doc/gcc-4.7/README.Bugs --enable-languages=c,c++,go,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.7 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.7 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --with-arch-32=i586 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.7.2 (Debian 4.7.2-5)

guitorri, 2015-05-06:
I don't have a GCC 4.7 available right now.... it could be a bug on the compiler:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=53096

Note that we are using GCC 4.8 and Clang on the Travis CI...

nvdl, 2015-05-06:
OK, will give GCC 4.8 a try.

guitorri, 2016-11-03:
Please reopen if still relevant.

closed 2016-11-03 by guitorri

nvdl, 2016-11-04:
If I recall correctly, it went away with 4.9.2.
Will see if there is any issue when making use of the stale PR.

Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.9/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --with-arch-32=i586 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.9.2 (Debian 4.9.2-10)