Qucs/qucs #251: Grid on/off setting saved in schematic file

nvdl, 2015-05-07:
Now, the setting value whether the grid should be on or off is saved in the schematic file.
Shouldn't it be an application setting?

Any rationale about that?

in3otd, 2015-05-08:
well, that was a decision of the original developers, many years ago, and they are no longer involved in the Qucs development, so I doubt we will ever know the original rationale...

Might have been as well in the Application Settings but I see no big reason to change it today, IMHO.

guitorri, 2015-05-08:
Isn't the saved property just to show/hide the grid? The grid size and the snapping to grid are set on the application, no?

in3otd, 2015-05-08:
Grid size and visibility are a property saved in the schematic.
Regarding the snap to grid, uhm, do we have a "snap to grid" setting ??

Actually I saw that changing grid size does change only the grid shown but the components are still placed on a 10x10 grid, so now I see one reason less to save the grid size in the schematic, since it does not constrain the component placement...

nvdl, 2015-05-08:
If I recall, I saw some 5x5 or 10x10 hard coded numbers in the code.
Now, I am using 5x5 grid to find a node and change the cursor (hard coded) until the settings are coded in.

I think these settings should be application wide and not hard coded.

With experience from another CAD, grid settings are saved application wide.
It might be "not so nice" experience if I disable or change the grid size and have to do that with every schematic I open later on. Same applies to a situation if I want to roll back to previous grid settings.

I think, visual settings should not be a part of the schematic.

closed 2018-07-19 by guitorri