Qucs/qucs #253: Right click menu for toolbar is missing text for the first three entries

nvdl, 2015-05-07:
File, Edit, View and Work show up properly but the other entries are missing text.
Can someone point me toward the source. I tried searching.

guitorri, 2015-05-08:
I never tried RMB on the menu and toolbar... cool.

The named ones are coming form here:
```
$ cd ~/git/qucs/qucs/qucs 
$ grep -rn QToolBar .
./qucs.h:44:class QToolBar;
./qucs.h:270:  QToolBar *fileToolbar, *editToolbar, *viewToolbar
./qucs_init.cpp:30:#include <QToolBar>
./qucs_init.cpp:886:  fileToolbar = new QToolBar(tr("File"));
./qucs_init.cpp:898:  editToolbar = new QToolBar(tr("Edit"));
./qucs_init.cpp:909:  viewToolbar = new QToolBar(tr("View"));
./qucs_init.cpp:917:  workToolbar = new QToolBar(tr("Work"));
```

The unnamed ones are the 3 docks: main left dock, octave dock and verilog-a message dock. It should be possible to add a name to these constructors:
```
$ cd ~/git/qucs/qucs/qucs
$ grep -rn 'new QDockWidget' .
./messagedock.cpp:59:    msgDock = new QDockWidget();
./qucs.cpp:225:  dock = new QDockWidget(this);
./qucs.cpp:365:  octDock = new QDockWidget();
```

nvdl, 2015-05-10:
Fixed by disabling the context menu and adding the entries to "View" menu.

guitorri, 2015-05-11:
Do you have a patch for this?

nvdl, 2015-05-11:
MessageDock is now inherited from QWidget so that it does not create a new dock necessarily. It can be inserted into the Octave dock or other common bottom dock.

Now, there is a separate dock for it.
The dock holding it is made visible or invisible through the view menu.

messagesDock = new QDockWidget();
messageDock = new MessageDock(this);

I think, it should work:
git cherry-pick 7a4edccafda533a7d4238b9cb9240d0a8963ce9e

But there may be some conflicts due to other already done changes.

closed 2015-06-27 by guitorri