Qucs/qucs #256: Gray code: Rename files and translatable strings

guitorri, 2015-05-26:
Some file names, classes, and a few translatable strings mention `grey` instead of `Gray`(from Gray code).

Spotted by @in3otd, see #182 

[ ] fix translatable strings without losing translations
[ ] fix file names and code

nvdl, 2015-05-30:
What about typos like these?

main.cpp:
QString orientation = "portraid";
--orin [portraid|landscape] set orientation (default portraid)

There is an issue with capatilization of strings as well.
lumped components -> Lumped components or Lumped Components.
no warning -> No warning

I changed these in the fork; the translations are broken now.

Sometimes the sentences are not consistent across the status text and tool tip text e.g. pop in and pop out of subcurcuit toolbar buttons.

guitorri, 2015-05-30:
The "portraid" case is not so much of a problem. It is probably a new string, not yet translated in the majority of the other languages. The capitalization is more of an issue. If translators do not review the changes, these will be marked as 'changed' and set to English by default. 
We don't have enough translators at the moment. We don't have a proper release cycle in place, with 'string freeze' to give time for translators to review the changes...
At least with Transifex the 'outdated' translation are not lost, they are kept on the translation memory. I will be easier for translators to spot small changes. We should reach out to get more people translating...

I think we should fix the strings as we find them... translators should step up. Please send a pull request with typos fixed then see when is best to merge.

nvdl, 2015-05-31:
Is it possible for the pull request to be merged partially?
I can send the pull request of the experimental branch where I merged in "spice4qucs" as well.

The merger can only take in the text strings and all can comment if the strings do not sound OK in the context.

The reason for this is that I have edited things that may need a communical consent to be a part of the main branch. That includes icon changes, text changes, GUI changes.
I am happy with the current state of it and have improvement plans when free.

Can google translator be of any help here?

This has been keeping me from sending a pull request.
So should I do that now?

in3otd, 2015-05-31:
yes, one can `cherry-pick` your comments and take only part of the overall changes. But it is not the usual way to merge the changes, it also puts more work on the person doing the merge.

I suggest that you do a new branch and use `git cherry-pick` to bring on that branch the changes you would like to submit. In this way it will be clearer which part you would like to change and if everything looks right it could be merged easily.

We (I) forgot to say that it is better to open a pull request early, even before you finish working on your changes, so everybody can review them. You can always update your pull request with new commits later, even change the code completely, before the merge.

nvdl, 2015-05-31:
As I see it, I can clone the Qucs master branch as one of the branches in my fork, apply the textual changes and create a pull request.

But this trivial task has set me on a longer journey of finding the git commands.

So once again I need someone's git foo to help me out if possible. :wink:

guitorri, 2015-05-31:
@nvdl, on Transifex you can use the Bing translator (Google translator token is not available for free anymore). There is a little button with a lightning bold, on the top right, that gives you a machine translated string. It can save you a lot of typing...

I see you found your way to a pull-request. You can continue doing anything to your branch, the pull-request will be updated accordingly. 

I am updating the wiki with tips for contribution....

nvdl, 2015-05-31:
Yes, I finally found my way into the pulls. :laughing:

I created an account on Transifex and can give Swedish a try.
But, for a while, fixing text and rephrasing some sentences is the plan.