Qucs/qucs #258: Strange error when no .qucs directory

leo-from-spb, 2015-05-31:
I've installed Qucs, then reveal a new strange empty directory .qucs (started with dot), and renamed it to qucs (without dots).
I started Qucs, made a schema, saved it to directory qucs (successfully) and then attempted to run it (F2). Qucs shown an error that it cannot save schema…

Creating a new directory .qucs (with dot) is a current workaround.

So, I see two problems here:

1. why does it tries to use directory .qucs when I asked to save schema into qucs;
2. in error message id doesn't show the file name and path, that can confuse users.

leo-from-spb, 2015-05-31:
P.S. version 0.0.18

nvdl, 2015-05-31:
.qucs is the working directory of Qucs where it saves the log, netlist etc for the simulation.
You can consider it as a temporary directory for the application.

leo-from-spb, 2015-05-31:
> .qucs is the working directory of Qucs
A strange requirement. Anyway, an application should check that such directory exists and write-available, and show a clear note what's happened.

P.S. what if there are other applications that put their stuff into ~/.qucs? ;)
I think it'd be better to ask to configure such paths at the first start, or when detected that the path doesn't exist or unavailable.

ra3xdh, 2015-06-01:
I don't see any problems here. Hidden directory in $HOME is standard behavior for all UNIX applications (For example ~/.mozilla --- for Firefox ~/.vim --- for Vim ~/.opera --- for Opera). All these applications uses these directories to store settings, cache, etc. All of these applications don't ask to configure this directory path.

closed 2015-06-01 by ra3xdh