Qucs/qucs #259: checker error, no actions defined: nothing to do

leo-from-spb, 2015-05-31:
Attempt to simulate has produced the following error:

> checker error, no actions defined: nothing to do

(see the attached screenshot). 
[image]

No details what kind of error is it, where it occurred and how to resolve it. 
Is a schema wrong? or something else?

So for me it's a show stopper (really, because I have no idea what happened and how to resolve).

leo-from-spb, 2015-05-31:
P.S. version 0.0.18

in3otd, 2015-05-31:
...please include or link the schematic you were trying to simulate, otherwise it might be difficult to guess what went wrong; anyway, I guess you did not place a simulation component (Simulations -> Components) to tell Qucs which kind of simulation to do...

leo-from-spb, 2015-05-31:
> please include or link the schematic you were trying to simulate

Hmm, I found no idea how to attach file to the issue. And I've already uninstalled the software and cannot take a screenshot of my schema.
I can send the file by e-mail, to which address?

> I guess you did not place a simulation component (Simulations -> Components) to tell Qucs which kind of simulation to do...

You're absolutely right! I didn't do it, I didn't know anything about that components.
In such case I'd show a question "There are no required components. Add them to the schema?", or even better adds all necessary components silently.

nvdl, 2015-05-31:
I agree, the error and warning system needs a bit of improvement especially for the beginners.

in3otd, 2015-05-31:
@nvdl, surely some error messages can be improved.
But adding a "Welcome" window at startup telling new users "*Please go to Help->Getting started to see how to use this SW*" and "*Under File->Examples you can find several example schematics ready to be used*" (and maybe together with some details on "What's new" for more experienced users) might be an even better idea.

@leo-from-spb , a simulation component is not required in every schematic, as it might be a subcircuit that is not supposed to be simulated alone. `Qucs` will not add any components silently, since it cannot guess what you would like to add and for doing what (even for simulations, there are 7 different types, which one would you like to do?).
To send a schematic, you can create a Gist in GitHub (top of the page, `Gist`link), you can use one of the many file hosting services (e.g. Dropbox) or you can simply paste the schematic file content (it is a text file)

leo-from-spb, 2015-05-31:
The file on the gist: 
https://gist.github.com/leo-from-spb/9fd1d2b22a4f6559522b#file-multivibrator-sch
Thank you for the suggestion.

> even for simulations, there are 7 different types, which one would you like to do?

May be to ask the user? (of course if no simulation components were added)

The problem that there are no ways for users even to guess that they have to add some special components in addition to usual details of schema.

nvdl, 2015-05-31:
+1 for home screen where you are shown some options; "new schematic with transient simulation", examples or a wizard.

leo-from-spb, 2015-05-31:
P.S. In short terms, this "show stopper" can be solved just to add a text message into the Messages window like "Please add simulation components". It allows a user to do something for resolving the problem and make the issue not so major.

nvdl, 2015-05-31:
Or the simulation button is greyed out until a simulation element is added.
It will remain grey and not allow a simulation element to be added to a subcircuit.

leo-from-spb, 2015-05-31:
> Or the simulation button is greyed out until a simulation element is added.

And show a 'tooltip' on hover this button why this is disabled.