Qucs/qucs #263: Usability Related Issues

nvdl, 2015-05-31:
[] - Keyboard based menu

nvdl, 2015-05-31:
**Keyboard based menu**

To enter a resistor, press p (place) and r (resistor).
To enter a capacitor, press p and c.

nvdl, 2015-05-31:
**Recently used elements**

Need to add a tab to the left tab widget to quickly select the recently used elements.

nvdl, 2015-06-03:
**Realistic model values**

Non-realistic value that might lead to convergence errors should not be allowed or the user should be warned against. For example an opamp with 0 ohm output impedance.

nvdl, 2015-06-04:
**Components stress testing**

Testing components that will be damaged after a specific voltage or current.
Their symbols will be updated after the tolerable threshold is crossed.

nvdl, 2015-06-06:
**FFT in cartesian like diagram without using the equation**

in3otd, 2015-06-06:
A very good thing will be to allow equations directly in the Graph trace expression, like being able to plot `dB(S[1,1])` directly, without having to define an equation on the schematic before. But I think this will need a **lot** of changes to the existing code...

in3otd, 2015-06-06:
Another, easier, thing that should be fixed, IMHO, is that the Graph Y-axis autoscaling is currently always done considering the full dataset, not only the portion actually selected and shown on the Graph. I mean, if you manually specify a range for the X axis, the Y axis will still be autoscaled considering also the points outside the selected range, which is often not what one really wants.

nvdl, 2015-06-06:
**dB(S[1,1])**
One way to calculate the new matrix, insert it into the existing list and make the diagram point to it.

**Y axis will still be autoscaled considering also the points outside the selected range**
Did not notice that as I was also specifying the y-range manually.

Nevertheless, the cartesian diagram brings the whole viewport to a freeze if there are many points.
I was thinking to create something based on OpenGL with points that fit the new screen pixels rather than plotting all the points on top of each other many times.

nvdl, 2015-06-06:
**Live graphing support**

Support for graph that updates as the simulation progresses.

nvdl, 2015-06-06:
**- [ ] Easy library sharing support**

1: Support for quick library sharing; copy link URL -> paste into Qucs.

2: Support for other library to Qucs library conversion in application (tools menu?).

3: Qucs online database which can be updated from menu.
The databse will help bring most needed models together rather than everyone making their own.

ra3xdh, 2015-06-08:
> Non-realistic value that might lead to convergence errors should not be allowed or the user should be warned against. For example an opamp with 0 ohm output impedance.

Sometimes non-realistic values are needed for debug. I used them in my circuits. I think it's not need to restrict them. Current design is more flexible. 

> Support for graph that updates as the simulation progresses.

I think this feature will be redundant.  Simulation in frequency domain is always sufficient fast, and it don't need to be dynamically shown.  If simlation in time-domain is slow, it means that something went wrong.  

> FFT in cartesian like diagram without using the equation

This will be a very useful feature. We can also implement PlotVs() function on display page. But it requires to move/ or duplicate a part of postprocessor form Qucsator to QucsGUI. As alternative we can implement a separated simple postprocessor like Ngnutmeg.

> Easy library sharing support

See my ideas on components library extension and library editor https://github.com/Qucs/qucs/wiki/Qucs-GSOC-2015-Ideas . We can merge these tasks.

nvdl, 2015-06-08:
**Sometimes non-realistic values are needed for debug. I used them in my circuits. I think it's not need to restrict them. Current design is more flexible.**

Is it possible to let the user know which paramater is causing the simulation problems?
Now, it just says "Jacobian".

**Support for graph that updates as the simulation progresses.**

The reason was to see if a simulation should be allowed to run or just stopped if something critical is seen. It will allow the user to decide the end of simulation time. Coupled to that, waveforms are not updated if simulation runs for a while and then crashes.

**As alternative we can implement a separated simple postprocessor like Ngnutmeg.**
We can give "Ngnutmeg" a try and see. No experience before.

**We can merge these tasks.**
I agree about merging.

nvdl, 2015-06-08:
- [ ] Support for simulating MCUs; AVR, ARM etc.

Comments are welcome.

nvdl, 2015-06-08:
- [ ] Y-Axis autoscaling issue in cartesian diagram

nvdl, 2015-06-08:
- [ ] Parametric run automation using a Python script

nvdl, 2015-06-08:
- [ ] **Sound capture and storing support**

Ability to read from a sound input device (not very important).
Ability to read from a sound file; can also be used to represent file-based voltage or current sources.
Ability to save waveforms as a sound file; ".wav" should be enough in the beginning.

in3otd, 2015-06-08:
for the file-based audio sources, have you already seen [this] (http://gareus.org/oss/spicesound/qucs) ? Quite old, but could still be useful...

guitorri, 2015-06-08:
This is turning into a _feature request_ list. It might be worth taking a look at the comments on the list of [requests on Sourceforge](https://sourceforge.net/p/qucs/feature-requests/?source=navbar).

For "Parametric test" there is some related stuff on the wiki that can be reused: [wiki/Related](https://github.com/Qucs/qucs/wiki/Related). Some other Python pieces might be found on our [regression suite](https://github.com/Qucs/qucs-test). 

All these ideas are great. Somehow we should pick the ones that we think that are doable and add them to the Roadmap. Hopefully in the future will be able to address them.

nvdl, 2015-06-09:
@guitorri: Yes, some of the features (audio and keyboard shortcuts) are coming from that list as I noted them down.

@in3otd: File-based audio sources looks like a good start. The license says "Stefan Jahn". Why did it not end up in the current repo then?

in3otd, 2015-06-09:
@nvdl I think he just carelessly used the existing files as a base for his modifications; AFAIK it was not Stefan who did that and I don't think he made the developers aware of his changes at that time. So if you plan to reuse that it will be better to explicitly ask his permission, even if it *appears* that the mods are under GPL...

ra3xdh, 2015-06-09:
>Now, it just says "Jacobian".

Unfortunately, "Jacobian singular..."  error may appear even if all parameters of components are realistic. For example it can take place for switching circuits at some time step in Transient analysis. So, non-realistic values don't cover all cases of failed simulations. It can even mislead user. 

> I agree about merging. How to do that? Move to wiki?

Yes, please open a new QEP https://github.com/Qucs/qucs/wiki/Qucs-Enhancement-Proposal It should cover the following issues:
1. Extend Qucs component library
2. Add library editor
3. Libraries sharing
4. Online libraries database and libraries updater
5. Also there was ideas on KiCAD libraries support and Qucs<-->KiCAD libraries converter.

ra3xdh, 2015-06-10:
> Support for simulating MCUs; AVR, ARM etc.

I have mixed feelings about MCU simulators. I think Qucs should not introduce this feature. It will be useful only for beginners, but useless for production. I was electronic devices developer before I became university lecturer. And I tried to use MCU simulators in production.  Sometimes simulator shows nonexistent bugs. It was faster to use development boards (like MSP430 launchpad, STM32-discovery) for devices prototyping. 

Now development kits such as MSP430-launchpad and STM32-discovery are very cheap. Even student can purchase them. These developments kits are thousand times more useful than simulators. 

Qucs has excellent frequency domain analog solver that is especially suitable for HF-circuits. I think we should follow this direction in Qucs development. Otherwise Qucs may become monster like Multisim. For example one of the best proprietary circuit simulators MicroCAP cannot simulate MCUs. I think Qucs should follow MicroCAP way.  Also my thoughts on this subject are here: https://github.com/Qucs/qucs/issues/193#issuecomment-73415080

You can use abandoned project KTechLab to implement digital simulator with additional MCU simulation features. KTechLab was one of the best open-source digital circuit simulators in past. Now this project seems to be dead. MCU s simulation feature can give this project a new live.

nvdl, 2015-06-10:
I agree, it is just a proposal and pursuing it depends on the amount of work and interest involved.
I would still say that it is aligned with the "Universal" word in Qucs.
There is a digital part in the simulator as well.

In the simplest term, we can make a generic device that connects to an external instance of a simulator i.e. gdb. Ability to simulate MCUs using an external interface will isolate Qucs being the main simulator. That possibility can still be used as a teaching tool though I do not see any problem in production. With such an interface, connection to a real hardware should also be possible if a simulator is missing.

nvdl, 2015-06-13:
- [ ] **Addition of sliders to transient and other simulation control diagrams**

Sliders are to be added to simulation control diagrams (and possibly others) for ease of updating the main control values.

nvdl, 2015-06-13:
- [ ] **Solar panel device or a similar source**

Addition of a source that has predefined I/V curve (loadable from a file) to simulate a solar panel or other sources. Possibly, there is already something similar implemented.

nvdl, 2015-06-13:
- [ ] **Auto apply component's parameter if text is changed and focus is lost**

Now, one needs to click on "Apply" for every parameter to be changed.
It will be better to apply the setting after a new parameter is selected for quickly updating many parameters. Need to have an "Undo" button if needed.

nvdl, 2015-06-16:
- [ ] **Flip X, Y axis checkboxes for cartesian (and other) diagrams**

Need to have checkboxes to easily flip the X and Y axes if needed (e.g. in D.C bias analysis where a resistor is swept from a high value to low value).

Now, one needs to enter the start and stop values manually (flipped) and have to readjust them if sweep range is changed.

nvdl, 2015-06-18:
- [ ] **Addition of slider to DC bias points selection window**

Addition of slider will make sweeping through the DC bias points easily.

nvdl, 2015-06-20:
- [ ] **Automatic timestep changes in transient analysis diagram**

The timestep is updated automatically when start/stop times are changed in the transient analysis diagram. Is that on purpose? I tend to keep the timestep fixed and modify start/stop time more often.

in3otd, 2015-06-20:
just for the sake of pickiness, you probably meant "component" instead of "diagram" above...

At present, when you change the start or stop times, the number of points is kept fixed and the timestep is changed. It's true it might make more sense to keep the timestep fixed and change the number of points, instead.

nvdl, 2015-06-20:
Compenent or element? I am lost. :astonished:

in3otd, 2015-06-20:
From a code point of view, the Transient Simulation is a component,
http://qucs.github.io/qucs-doxygen/qucs/classTR__Sim.html, and every component is an Element.
The Diagrams are Elements but not Components http://qucs.github.io/qucs-doxygen/qucs/classDiagram.html .

Then, the presentation in the GUI is slightly different...

closed 2018-07-19 by guitorri

Foadsf, 2019-02-05:
There is also [simavr](https://github.com/buserror/simavr)