Qucs/qucs #265: Remove unnecessary null pointer checks

elfring, 2015-06-01:
[Extra null pointer checks are not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for NULL before delete p?") like the following.
* [circuit](https://github.com/Qucs/qucs/blob/e2448cde61fb415a857c526cf059019fd1e49d6e/qucs-core/src/circuit.cpp#L183 "Destructor for circuit")
* [hbsolver](https://github.com/Qucs/qucs/blob/51e4d23548e7d6c2f9342c98b2070b7150a3419c/qucs-core/src/hbsolver.cpp#L84 "Destructor for hbsolver")
* [spline::realloc](https://github.com/Qucs/qucs/blob/e2448cde61fb415a857c526cf059019fd1e49d6e/qucs-core/src/spline.cpp#L143 "spline::realloc function")

guitorri, 2015-06-01:
@elfring thank you for reporting. The above patch should fix the issue.

elfring, 2015-06-01:
Thanks for such an update.

Would it be safer to reuse smart pointers also for your software?

guitorri, 2015-06-01:
It will be definitely safer to use smart pointers. 

We should discuss that, but in the end someone needs to step up and do the refactoring. Please open another issue if to bring the discussion up... it is not *my* software after all... :smile:

Are you using a tool o spot these non-conformances?

elfring, 2015-06-01:
A limited source code analysis tool can be a search interface like that from GitHub.

guitorri, 2015-06-01:
Interesting, I seldom use the search box. Nice hint. Thank you again for taking the time to report issues!

closed 2015-06-20 by in3otd

elfring, 2015-06-20:
Thanks for [your source code improvement](https://github.com/Qucs/qucs/commit/8c2dd9e909f0554ee0d1e4a305e54ad68191ed8e "Deletion of unnecessary pointer checks").

commit 8c2dd9e9 (2015-06-20): Merge branch '265-remove-null-checks' from PR #267

Closes #265