Qucs/qucs #272: Calculate DC bias with Sweep is broken

in3otd, 2015-06-08:
[This schematic] (https://gist.github.com/in3otd/82c466d43522a565cb8b) using a Sweep with a DC simulation crashes the Qucs GUI when running `Calculate DC bias` (`F8`) . Simpler schematics often run, but the small dialog allowing to select the sweep run to use for showing the DC bias is not shown. Was apparently broken also in 0.0.18 (??).

nvdl, 2015-06-09:
qucp.cpp:

SweepDialog *Dia = new SweepDialog((Schematic*)sim->DocWidget);

is causing the fault.

I checked and "sim->DocWidget" is not NULL.

"SweepDialog::SweepDialog(Schematic *Doc_)" seems to cause this.

in3otd, 2015-06-09:
yes, I saw the `gdb`backtrace... I was a bit puzzled since the only recent changes are 0a2f1a7166558281065e391c78a00e2e9f8346b5 and a09e00cb726544dfa0e2a1cff44c31485b2b9096 ; moreover it seems to crash already on 0.0.18...

nvdl, 2015-06-09:
Will check it later.

What about:
"delete pGraph;" line?

nvdl, 2015-06-13:
Another crash example:

https://gist.github.com/nvdl/5ebe56ddebba5986ba95

smile5, 2015-06-16:
With qucs 0.0.18, it crash with the two schematics.

If that can help:
But if in the first "dc_bias_crash.sch" i f you rename the parameter, from "Vin" to another name, it doesn't crash...
For the second, if you add a label, and if you modify the sweep param from "R4" to "R_var" modifying also the R4 value all is correct..

Perhaps a bad use of sweep parameter ?

nvdl, 2015-06-16:
**But if in the first "dc_bias_crash.sch" i f you rename the parameter, from "Vin" to another name, it doesn't crash...**

I renamed it to "abc" and it still crashes.

smile5, 2015-06-17:
I retry, redownloaded example, modify the parameter value to "abc", save the file, and press F8, and don't crash with version 0.0.18, but the small window to choose the DC point not appear all the time, only the first time...

nvdl, 2015-06-18:
OK, I might have again missed something.
Can you check the pull request above? It fixes stuff on my side.

guitorri, 2015-06-18:
This is the relevant part of the crash report I get with the first schematic:
```
0   qucs                          	0x000000010ca7c097 mySpinBox::textFromValue(int) const + 71 (sweepdialog.cpp:49)
1   QtGui                         	0x000000010d1d5bfa QSpinBoxPrivate::textFromValue(QVariant const&) const + 58
2   QtGui                         	0x000000010d13bb81 QAbstractSpinBoxPrivate::updateEdit() + 273
3   QtGui                         	0x000000010d1d6194 QSpinBox::setSingleStep(int) + 68
4   qucs                          	0x000000010ca7bfca mySpinBox::mySpinBox(int, int, int, double*, QWidget*) + 122 (sweepdialog.cpp:37)
5   qucs                          	0x000000010ca7c047 mySpinBox::mySpinBox(int, int, int, double*, QWidget*) + 55 (sweepdialog.cpp:42)
6   qucs                          	0x000000010ca7c624 SweepDialog::SweepDialog(Schematic*) + 1108 (sweepdialog.cpp:101)
7   qucs                          	0x000000010ca7d5ad SweepDialog::SweepDialog(Schematic*) + 29 (sweepdialog.cpp:114)
8   qucs                          	0x000000010c63c69e QucsApp::slotAfterSimulation(int, SimMessage*) + 254 (qucs.cpp:2058)
```

guitorri, 2015-06-18:
@nvdl your pull request seems to fix the crash. I left some comments on the PR.

guitorri, 2015-06-23:
This passing around of pointers is hurting my brain... has anyone looked at it?

guitorri, 2015-06-24:
Another fix attempt and more discussion in  PR #282.

closed 2015-06-26 by guitorri