Qucs/qucs #281: Should the "Update schematic path" popup be removed?

in3otd, 2015-06-23:
When one saves all the open tabs using "Save all" or uses "Save As" a modal message box telling that ""The schematic file path has been refreshed" pops up and forces the user to press the Ok to continue.
I fail to see why the user should be informed about this in a way which seems to me too intrusive. I think the search path could well be updated without informing the user. Or if it is really important we might just show a message on the status bar. 
What do **you** think?
@crobarcro , what was/is the purpose of the modal dialog?

Of course this is about removing just the modal dialog, the search path feature will not be touched.

crobarcro, 2015-06-23:
To be honest, I think I've left this in by accident while testing. Feel free to remove it!

guitorri, 2015-06-23:
A message on the status bar seems fair...

nvdl, 2015-06-23:
Message on the status bar. +1

in3otd, 2015-06-23:
ok, I will implement the status bar message...

closed 2015-06-26 by guitorri