Qucs/qucs #284: Reading simulation output data efficiently in terms of speed and memory usage

nvdl, 2015-06-25:
Coupled to #282

nvdl, 2015-06-25:
- [ ] Deciding about the output data format of "qucsator"

nvdl, 2015-06-25:
- [ ] **Deciding about the output data format of "qucsator"**

The current format is not so efficient in terms of space and possibly speed if a lot of points are written at a fast rate.

Need to move to another data output format for "qucsator".
Or, as a first step, convert the current data format to HDF5 (or other) after the simulation is complete and use the new file.

nvdl, 2015-06-25:
- [ ] Using a class to read the data file

Need to write a new class that handle all the simulation data requirements of the software.

nvdl, 2015-06-25:
- [ ] **Remove duplicate code**

All the code relevent to the loading of siumlation data should go the new proposed class.
Duplicate functions should be removed.

felix-salfelder, 2015-06-25:
@nvdi: the data format does not matter, can be changed any time. after the classes for data sources have settled, we should do the same for simulators. qucsator is just one of them, and they should be treated alike (by the GUI). THEN, qucs would just do (simplified for readability)

    simulator->init();
    for(item in schematic){
      simulator->instanciate(item);
    }
    simulator->run();
    outputData=simulator->results().

here you see, that the path between the simulator and the frontend is an implementation detail of the simulator class, not of the frontent.

PS: don't know what HDF5 is, but it should be easy to add. what we need is

    SimOutputData const* load_data(filename){
         parser P(filename);
         for (F in fileFormats){
             if F->understands(P) return F->parse(P);
         }
         return NULL
    }

(and the fileFormats are implemented as plugins). please do not change the .dat file format before the parser is completely implemented as a plugin.

nvdl, 2015-06-25:
I agree, I added steps to the issues so we can follow the flow of changes.

felix-salfelder, 2015-06-25:
on **Using a class to read the data file**
this is blocking my transition currently: #285

guitorri, 2015-06-25:
@nvdl I believe `qucsator` only writes to its plain text dataset format. `qucsconv` can convert to other [formats](https://github.com/Qucs/qucs/wiki/Qucs-Command-Line-Tools#qucsconv) (like csv, mat).

I guess the checklist should be gathered on the topmost comment. Then the issue tracker shows how many tasks are done or not.

nvdl, 2015-06-25:
_I believe qucsator only writes to its plain text dataset format._
That is not an issue for a while but I feel the disk I/O can be optimized to speed up the simulation if there is less data to write. The floating point value is now consuming 21 bytes in the text format.

I ended up with a file size of 1GB+ quite easily.

nvdl, 2015-06-25:
@felix-salfelder
_because there are only a few pixels on the screen, and it also makes sense to select a subset of the data for plotting purposes_
I totally agree with you. This was also my plan. The whole schematic slows down when there are more data points.

What do you mean by "marker code"?

felix-salfelder, 2015-06-25:
> What do you mean by "marker code"?

have a look at `Marker::moveUpDown` and `LeftRight` in `diagrams/marker.cpp`, and how it does a linear search on the (full) data set using `double*`...

nvdl, 2015-06-25:
- [ ] Keep code consistent

Now "diagram.cpp" is hosting code for "Graph" class that will better be placed in "graph.cpp".

nvdl, 2015-06-25:
- [ ] ".dat" to HDF5 converter (step 1)

nvdl, 2015-06-25:
- [ ] X-simulator output to HDF5 converter (step X)

nvdl, 2015-06-25:
- [ ] Configure "qucsator" to write out HDF5 directly (step X)

felix-salfelder, 2015-06-26:
**".dat" to HDF5 converter (step 1)**

no. don't convert. write HDF5 in the first place. conversion is inefficient.

felix-salfelder, 2015-06-26:
**Keep code consistent (step 1)**

i chose to not move it immediately, as this makes it harder to track. need to fetch up.
this is the last thing to to (once i am/we are sure about it).

felix-salfelder, 2015-06-26:
**X-simulator output to HDF5 converter (step X)**

no. dont convert. this is inefficient. read other simulator output as well.

there will be no way to read data from a simulator directly, if all qucs understands is HDF5.

nvdl, 2015-06-26:
_no. don't convert. write HDF5 in the first place. conversion is inefficient._

Yes, it is inefficient but is one shot. If later lookups from HDF5 are faster and convenient, conversion is worth it.

_there will be no way to read data from a simulator directly, if all qucs understands is HDF5._

In the last step, we should modify "qucsator" to write to HDF5 and some other data formats (if needed).
Have you looked into the current ".dat" file? It is just plain text; good for debugging but not efficient for simulation results.

in3otd, 2015-06-26:
can we add some number to the discussion? Where is the current bottleneck? Reading the file or plotting the data? How much time these operations take with the current code? How could we say we improved this in the new implementation if we have no data to compare with?
Can you define some "typical" schematic/data to use as benchmark? My simulations usually generate small `.dat` files; are we sure the new format will be efficient also for this case?

felix-salfelder, 2015-06-26:
@in3otd primarily, we have a structural problem. `qucs-gui` does only read from .dat files, this will stop working, once the data is contained in a hdf5 file. the conversion requires space and time. both approximately linear in the size of the data set. the current implementation of the dat file parser is quadratic in the number of nodes. doing no conversion is constant in time, it should be easy to feel the difference.

another problem is that plotting (and moving markers) should only happen on a (visible) subset of the data. this visibility is constrained by the plot boundaries and the number of pixels on the screen. this is not implementable currently, e.g. the marker code directly reads from the dat file image.

> are we sure the new format will be efficient also for this case?

yes. if you don't reorder data after it pops out of the simulator(-core), this will even be faster on small data sets. my guess is that HDF5 does not require reordering (can somebody check please?), but i'm not fixated to HDF5.

in3otd, 2015-06-26:
yeah, the questions were meant to provoke some discussions...
I fully agree with supporting **also** HDF5, both as output of `qucsator` and also as input of Qucs. But the reason for me is **versatility**. Support to more *standard* data formats is good for the users. Text-based data formats have been the standard (on unix, at least) for ages, and won't go away tomorrow. For the same reason, I will be happy to see also FSDB (used by some commercial simulator) supported (open standard ? GTKwave supports it) and maybe also the *tabular trace file format* defined for the SystemC AMS (duh! a text-based data format!)
Now, if we say "we need HDF5 for speed reasons" I would like to see some more reasoning about that.
Good to see some comments on the current algorithm complexity, we are getting closer to numbers...
You said "the current implementation of the dat file parser is quadratic in the number of nodes." which we agree is not a good thing. Do we **need** to change the data format to improve on this or can we "just" code a better parser for `qucsdata`? (I do not know the answer). Also "another problem is that plotting [...] directly reads from the dat file image"; again this is a problem of the implementation rather than the data format. We could surely have the same issue with the HDF5 format if we try hard enough to write bad code.

I am not saying that a text data format is faster to handle than a binary one, but I do not remember having a speed issue when plotting data for my typical usage, so I would like to have some numbers and examples from others who do see this issue.

felix-salfelder, 2015-06-26:
i do not care about the speed of the default file format. cards on the table, i "just" want to implement an interface to simulators, so i can use gnucap as a backend, in a sane and *efficient* way -- maybe interactively, but that is another story.

this mandates a more generic interface to simulator data in general. it is important to me that the interface *can* be used for other purposes, such as improving the dat parser, adding a HDF5 parser or turning `qucsator` into a library (and i am eager to assist you with the implementation).

> Do we need to change the data format to improve on this or can we "just" code a better parser for qucsdata?

while you can read in .dat files in one piece (see qucsdata.py), it's not possible to generate them efficiently. the simulator has to collect result data and then put together the chunks in a different order. writing out data as the computation proceeds is much more efficient and clearly simpler than that.

felix-salfelder, 2020-09-26:
this has been addressed (and drafted) elsewhere. someone will have to patch qucsator to make it faster. qucs (the gui) will hopefully allow adding support for other data file formats.

closed 2020-09-26 by felix-salfelder