Qucs/qucs #295: Issue with gperf 3.0.4

guitorri, 2015-07-02:
While doing some testing on Windows, the build broke down because I have gperf-3.0.4 in the path. It works just fine with gperf-3.0.1.

Using MSYS2/Autotools. 

```
./gperfappgen.exe > gperfapphash.gph
/usr/bin/gperf -I -m 8 gperfapphash.gph | sed -e 's/{""},/{"",0},/g' > gperfapphash.cpp
gperfapphash.gph:8: warning: junk after %% is ignored
, defaulting to Cage option C++
gperfapphash.gph:6: junk after declaration
/bin/sh ../libtool --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..  -I../src/math -I../src/components -I../src/interface -I.   -g -               O2 -std=gnu++11 -O0 -pipe -fno-exceptions -fno-check-new -W -Wall -MT equation.lo -MD -MP -MF .deps/equation.Tpo -c -o equation.lo equation               .cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -I../src/math -I../src/components -I../src/interface -I. -g -O2 -std=gnu++11 -O0 -pipe -fno               -exceptions -fno-check-new -W -Wall -MT equation.lo -MD -MP -MF .deps/equation.Tpo -c equation.cpp  -DDLL_EXPORT -DPIC -o .libs/equation.o
equation.cpp: In member function 'int qucs::eqn::application::evalTypeFast()':
equation.cpp:807:29: error: 'gperfapphash' has not been declared
     struct appindex * idx = gperfapphash::get (key, strlen (key));
                             ^
equation.cpp:811:48: error: invalid use of incomplete type 'struct qucs::eqn::application::evalTypeFast()::appindex'
         application_t * app = &applications[idx->index];
                                                ^
equation.cpp:807:12: error: forward declaration of 'struct qucs::eqn::application::evalTypeFast()::appindex'
     struct appindex * idx = gperfapphash::get (key, strlen (key));
            ^
Makefile:661: recipe for target 'equation.lo' failed
```

guitorri, 2015-09-18:
Maybe the issue is related with line ending instead of gperf version? :confused: 
https://bugs.webkit.org/show_bug.cgi?id=141507

I hit the problem when using gperf provided with MSYS2.
To work around the issue I simply put the [gnuwin32 gperf version](http://gnuwin32.sourceforge.net/packages/gperf.htm) in front of the PATH....

guitorri, 2016-02-03:
Similar issue when cross compiling with mingw (Mac to Wine) to build binary package.

```
~/git/build_qucs $ gperf --version
GNU gperf 3.0.3
```

Error:
```
equation.cpp: In member function 'int qucs::eqn::application::evalTypeFast()':
equation.cpp:807:29: error: 'gperfapphash' has not been declared
     struct appindex * idx = gperfapphash::get (key, strlen (key));
                             ^
equation.cpp:811:48: error: invalid use of incomplete type 'struct qucs::eqn::application::evalTypeFast()::appindex'
         application_t * app = &applications[idx->index];
                                                ^
equation.cpp:807:12: error: forward declaration of 'struct qucs::eqn::application::evalTypeFast()::appindex'
     struct appindex * idx = gperfapphash::get (key, strlen (key));
```

Workaround, install another gperf into wine.

felix-salfelder, 2016-02-03:
hmm. what does it do? can we make it optional?

guitorri, 2016-02-03:
don't know how easy it is to remove it... 
it is used to generate some sort of lookup table for the equation system. I haven't been in those parts of the code in a while...

felix-salfelder, 2016-02-03:
are you sure that this is not related to a mixed up gperf? (i mean something like gperf.4 in the path, but .1 headers in /whatever/include?)

(just a shot in the dark)

guitorri, 2016-02-03:
I will have a look around.

For the moment I will add a similar option to --with-gperf, so I can workaround the issue for this release...

I will try to reproduce/investigate further on Windows/MSYS2 where the issue first showed up.

guitorri, 2016-02-03:
hum.. somewhere along the way:
```
gperf -I -m 8 gperfapphash.gph | sed -e 's/{""},/{"",0},/g' > gperfapphash.cpp
gperfapphash.gph: The input file is empty!
```

in3otd, 2016-02-10:
indeed by converting the `\r\n` to `\n` (with `dos2unix`) in `gperfapphash.gph` also `gperf` 3.0.4 works fine. (slowly trying to build under Windows)

guitorri, 2016-02-10:
Thanks! This is part of now #462

in3otd, 2016-02-10:
the "strange" thing is that `gperfapphash.gph` is generated by redirecting the output of [`gperfappgen`] (https://github.com/Qucs/qucs/blob/master/qucs-core/src/gperfappgen.cpp) to a file, so even if it uses `\n` as EOL it gets automatically converted to `\r\n`. I do not know if there is a way to avoid this, my understanding is that this is a Windows "feature" when writing text files...

felix-salfelder, 2016-02-11:
sounds like a windows bug. will this work

    #if WINDOZE
    FILTER=| dos2unix
    #endif 

     somefile: gperfappgen gperfwhatnot
     \t./gperfappgen < gperfwhatnot $FILTER > somefile

?

in3otd, 2016-02-11:
yep, this seems really a WIndows/Cygwin/MSYS "feature" actually :grin:
https://cygwin.com/cygwin-ug-net/using-textbinary.html

it seems that one cannot even be sure that piping and redirecting are doing the same conversion... So above it could be that the conversion to `\r\n` is done after the filter, by the redirection. Will try later to see what happens.
Still will be interesting to know why the behavior was changed between gperf-3.0.1 and 3.0.4

felix-salfelder, 2016-02-11:
ok then maybe

    #if WINDOZE
    FILTER=dos2unix somefile
    #endif 
    somefile: gperfappgen gperfwhatnot
    \tcommand...
    \t$(FILTER)

sorry, I cannot test stuff like this, for obvious reasons `:/`

in3otd, 2016-02-11:
@felix-salfelder  FYI, your [first solution] (https://github.com/Qucs/qucs/issues/295#issuecomment-182750416) seems to work here

guitorri, 2016-02-12:
Fixed with #465

closed 2016-02-12 by guitorri