Qucs/qucs #296: Regression on some EDD components/equations

guitorri, 2015-07-02:
@MikeBrinson reports that the schematic/project below is crashing qucsator.
He also reports that removing the `TAU*I3` from the equation `Q3 = DCS*V3 + TAU*I3` inside the subcircuit avoids the crash.

The model is working with 0.0.18.

Project: https://gist.github.com/guitorri/bdcde7b5527bc32b92a5

@ra3xdh got this backtrace:
```
There is GDB backtrace of crash:
(gdb) bt
#0  0x00007ffff66c4a58 in main_arena () from /lib64/libc.so.6
#1  0x00007ffff728521f in qucs::eqn::assignment::add (this=0x633b20, f=0x634060) at equation.cpp:568
#2  0x00007ffff737c55f in eqndefined::initModel (this=0x61db00) at eqndefined.cpp:249
#3  0x00007ffff737bb5d in eqndefined::initDC (this=0x61db00) at eqndefined.cpp:67
#4  0x00007ffff726b880 in qucs::dcsolver::init (this=0x61f3b0) at dcsolver.cpp:186
#5  0x00007ffff726b1fd in qucs::dcsolver::solve (this=0x61f3b0) at dcsolver.cpp:76
#6  0x00007ffff72824f3 in qucs::parasweep::solve (this=0x61dec0) at parasweep.cpp:170
#7  0x00007ffff72824f3 in qucs::parasweep::solve (this=0x61e7e0) at parasweep.cpp:170
#8  0x00007ffff725dbb1 in qucs::net::runAnalysis (this=0x60c190, err=@0x7fffffffdd8c: 0) at net.cpp:264
#9  0x00000000004032d6 in main (argc=7, argv=0x7fffffffe318) at ucs.cpp:251
```

I got a different traceback, with the warning:
```
qucsator(74923,0x7fff7c5b8310) malloc: *** error for object 0x7fe36bc26060: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
``` 
I tried reverting f9dbd6b3 but no luck so far.

in3otd, 2015-07-02:
When I first checked #267 I thought I saw some `ptr=NULL` that was removed by the patch, but then I was not able to find it again...
I guess you are already bisecting, in case you do not have around a script for automatic bisection on `qucs-core` you can try to use [this] (https://gist.github.com/in3otd/053641a2bb24d7303c66) . I do not guarantee is it still working (used it quite some time ago), but might be useful if you do not have already something similar around

in3otd, 2015-07-02:
ok, `git bisect` says that 9bd368d15a6e2eb17be3175f11de2b7b8fe306ce is the bad commit, can someone take a look at it ?

guitorri, 2015-07-03:
Commit 9bd368d is/was crashing for another reason, see below. I think this particular issue was fixed in later commits... I will try to bisect again. @in3otd tanks for the script!

```
[(9bd368d...)]~/git/qucs/qucs-core $ lldb -- /Users/guitorri/local/qucs/test/bin/qucsator -i /Users/guitorri/qucs_area/bugs/netlist.txt -o 
[...]
Process 34090 stopped
* thread #1: tid = 0x4617d0, 0x0000000100060774 libqucs.0.dylib`std::list<qucs::analysis*, std::allocator<qucs::analysis*> >::begin(this=0x0000000000000000) + 20 at stl_list.h:583, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x0000000100060774 libqucs.0.dylib`std::list<qucs::analysis*, std::allocator<qucs::analysis*> >::begin(this=0x0000000000000000) + 20 at stl_list.h:583
   580 	       */
   581 	      iterator
   582 	      begin()
-> 583 	      { return iterator(this->_M_impl._M_node._M_next); }
   584 	
   585 	      /**
   586 	       *  Returns a read-only (constant) iterator that points to the
(lldb) bt
* thread #1: tid = 0x4617d0, 0x0000000100060774 libqucs.0.dylib`std::list<qucs::analysis*, std::allocator<qucs::analysis*> >::begin(this=0x0000000000000000) + 20 at stl_list.h:583, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  * frame #0: 0x0000000100060774 libqucs.0.dylib`std::list<qucs::analysis*, std::allocator<qucs::analysis*> >::begin(this=0x0000000000000000) + 20 at stl_list.h:583
    frame #1: 0x000000010005fab9 libqucs.0.dylib`qucs::net::sortChildAnalyses(this=0x0000000100c08160, parent=0x0000000100c13b60) + 57 at net.cpp:350
    frame #2: 0x000000010005f75f libqucs.0.dylib`qucs::net::orderAnalysis(this=0x0000000100c08160) + 527 at net.cpp:334
    frame #3: 0x000000010005f2fe libqucs.0.dylib`qucs::net::runAnalysis(this=0x0000000100c08160, err=0x00007fff5fbff320) + 302 at net.cpp:250
    frame #4: 0x0000000100002ae4 qucsator`main(argc=5, argv=0x00007fff5fbff968) + 4740 at ucs.cpp:250
    frame #5: 0x0000000100001854 qucsator`start + 52
(lldb) 

```

in3otd, 2015-07-03:
ah, didn't check the segfault cause, I set the bisect range from 0.0.18 to HEAD, to broad likely.

Note that you need to modify the script, at present it check for NaNs in the output file, to capture segfaults you will need to check for the `qucsator`output code , of course

guitorri, 2015-07-03:
This one might be hard to bissect, just like #248 was. There too commits that are either failing for unrelated reasons or do not build at all on my system...

guitorri, 2015-07-03:
I think this is the culprit:
https://github.com/Qucs/qucs/blob/master/qucs-core/src/equation.cpp#L575
This line should not be commented out... Shame on me (6a073cf48).

in3otd, 2015-07-03:
yep, also the `bisect` here now agrees :grin: 
Maybe add a project with this circuit to `qucs-test`, at present there is just one test for HB there...

guitorri, 2015-07-03:
It should be fixed now. A test was added to qucs-test.
Yes, we need to improve the test coverage... and figure out a way to test the GUI part as well.

closed 2015-07-03 by guitorri