Qucs/qucs #300: QucsActiveFilter results on Windows

in3otd, 2015-07-03:
When running `QucsActiveFilter` with the default parameters, I get different results on Linux and Windows.

To reproduce:
- open the `QucsActiveFilter` tool
- press the button `Calculate and copy to clipboard` (i.e. leave all the parameters at the default)
- on Linux it runs fine. On Windows it fails

This is the output of the Filter Calculation Console:

- On Linux:
````
Filter order = 14

Poles list Pk=Re+j*Im
-0.111964 + j*0.993712
-0.330279 + j*0.943883
-0.532032 + j*0.846724
-0.707107 + j*0.707107
-0.846724 + j*0.532032
-0.943883 + j*0.330279
-0.993712 + j*0.111964
-0.993712 + j*-0.111964
-0.943883 + j*-0.330279
-0.846724 + j*-0.532032
-0.707107 + j*-0.707107
-0.532032 + j*-0.846724
-0.330279 + j*-0.943883
-0.111964 + j*-0.993712

Part list
Stage# C1(uF) C2(uF) R1(kOhm) R2(kOhm) R3(kOhm) R4(kOhm)  R5(kOhm) R6(kOhm)
     1    62.680pF 10000.000pF   284.295   284.295   142.148     0.000     0.000     0.000
     2   545.421pF 10000.000pF    96.376    96.376    48.188     0.000     0.000     0.000
     3  1415.291pF 10000.000pF    59.829    59.829    29.915     0.000     0.000     0.000
     4  2500.000pF 10000.000pF    45.016    45.016    22.508     0.000     0.000     0.000
     5  3584.709pF 10000.000pF    37.593    37.593    18.797     0.000     0.000     0.000
     6  4454.579pF 10000.000pF    33.723    33.723    16.862     0.000     0.000     0.000
     7  4937.320pF 10000.000pF    32.032    32.032    16.016     0.000     0.000     0.000

Filter calculation was sucessfull
````
- on Windows
````
	

    Filter order = 14
     
    Poles list Pk=Re+j*Im
    -0.111964 + j*0.993712
    -0.330279 + j*0.943883
    -0.532032 + j*0.846724
    -0.707107 + j*0.707107
    -0.846724 + j*0.532032
    -0.943883 + j*0.330279
    -0.993712 + j*0.111964
    -0.993712 + j*-0.111964
    -0.943883 + j*-0.330279
    -0.846724 + j*-0.532032
    -0.707107 + j*-0.707107
    -0.532032 + j*-0.846724
    -0.330279 + j*-0.943883
    -0.111964 + j*-0.993712
     
    Part list
    Stage# C1(uF) C2(uF) R1(kOhm) R2(kOhm) R3(kOhm) R4(kOhm)  R5(kOhm) R6(kOhm)
         1    62.680pF 10000.000pF       nan       nan       nan     0.000     0.000     0.000
         2   545.421pF 10000.000pF       nan       nan       nan     0.000     0.000     0.000
         3  1415.291pF 10000.000pF       nan       nan       nan     0.000     0.000     0.000
         4  2500.000pF 10000.000pF    45.009    45.009    22.511     0.000     0.000     0.000
         5  3584.709pF 10000.000pF       nan       nan       nan     0.000     0.000     0.000
         6  4454.579pF 10000.000pF    33.717    33.717    16.865     0.000     0.000     0.000
         7  4937.319pF 10000.000pF    32.026    32.026    16.019     0.000     0.000     0.000
     
    Filter calculation terminated with error
````

any idea ?

ra3xdh, 2015-07-04:
@in3otd, There is no bug it `qucsactivefilter`. I think that it is bug in MinGW. Are you using `cmake` to build? DEBUG mode is always enabled in `CMake` build system. Try to disable DEBUG mode and this bug disappears. I think this is connected with `floats` processing in MinGW. I have just rebuilt `qucs-activefilter` with disabled DEBUG mode with MnGW-4.8.2  and Qt-4.8.6 on my WinXP-32bit VirtualBox. This bug disappears. Screenshot is attached. I think we can ignore this, because `relase` version will not have this bug (DEBUG is disabled).
[image]

in3otd, 2015-07-04:
uhm, might be a bug in `MinGW` but it reminds me of #186 in some way (there changing the compile options made the bug appear and disappear). In any case I think we should try to find out precisely where the bug comes from, otherwise it might happen also when not in DEBUG mode, if we modify the code one day.

in3otd, 2015-07-14:
...it's not actually a bug in `MinGW`, just a small [error of the third kind] (https://books.google.com/books?id=t1LCAgAAQBAJ&lpg=PR11&ots=hAFIOix46_&dq=%22errors%20of%20the%20third%20kind%22&pg=PR11#v=onepage&f=false). More on this later :grin:.

guitorri, 2015-09-17:
Is this fixed? Can we close the issue?

ra3xdh, 2015-09-17:
@in3otd and I prepared fix for this. There is #312 that is not merged yet. I need further feedback from @in3otd for https://github.com/in3otd/qucs/pull/2. Then #312 could be merged and #300 closed.

in3otd, 2015-09-17:
@ra3xdh will try to take a look and merge your PR soon...

PR #312 (2015-09-28): Issue 300
This fixes #300.
Actually the fix is just the first commit, 08e0e70f8287316f373e70461b86b327594d83b5, the others are just small fixes here and there. The last one, 3b42993761ad6b535d87909ac1f60b3acf7f3e90 , changes a bit the window layout to make it more compact, Looks nicer to me, YMMW, leave it out if you prefer.

Some general notes, for future developments:
- maybe it would have been better to use `double` instead of `float` to hold the intermediate results (poles/zeros, etc.), this might have avoided the numerical errors above (did not actually try)
- in [`mfbfilter.cpp`] (https: [... cut]
closed 2015-09-28 by guitorri