Qucs/qucs #305: graph crash while opening schematic.

guitorri, 2015-07-10:
Qucs is crashing while opening the following project (works with 0.0.18):
https://gist.github.com/guitorri/e3bec9dd1f1bf1f4a175
(dataset included)

On the terminal it reports:

```
Assertion failed: (x>=0), function setScrX, file graph.cpp, line 317.
```

Crash report:

```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8dbb5866 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff8439335c pthread_kill + 92
2   libsystem_c.dylib             	0x00007fff817dbb1a abort + 125
3   libsystem_c.dylib             	0x00007fff817a59bf __assert_rtn + 321
4   qucs                          	0x000000010f911ffc Graph::ScrPt::setScrX(float) + 92 (graph.cpp:317)
5   qucs                          	0x000000010f912158 Graph::ScrPt::setScr(float, float) + 40 (graph.cpp:334)
6   qucs                          	0x000000010f92b8d2 Diagram::calcCoordinateP(double const*, double const*, double const*, std::__1::__wrap_iter<Graph::ScrPt*>&, Axis const*) const + 146 (diagram.cpp:1959)
7   qucs                          	0x000000010f92af8a Diagram::calcData(Graph*) + 1610 (diagram.cpp:524)
8   qucs                          	0x000000010f92e17b Diagram::updateGraphData() + 299 (diagram.cpp:760)
9   qucs                          	0x000000010f92c7aa Diagram::loadGraphData(QString const&) + 1002 (diagram.cpp:717)

```

guitorri, 2015-07-10:
@felix-salfelder can you take a look at this? Most likely another corner case...

felix-salfelder, 2015-07-12:
my assertion is too optimistic. the negative coordinates appear to be reserved for control tokens (`BRANCHEND`, `GRAPHEND`). they really are not, in some corner cases.
the `STROKEEND` has become obsolete with the qt painting #297, the others will follow.
i will provide an intermediate quick fix (remove the assertion?), if i don't manage to untangle this soon.

felix-salfelder, 2015-07-16:
#313

guitorri, 2015-07-18:
Merged. Thanks!

closed 2015-07-18 by guitorri