Qucs/qucs #307: Edit Component Properties: hard to read property description

guitorri, 2015-07-13:
Issues:
- [ ] the Description on the right portion should not be clipped with `...`, break lines instead
- [ ] the column width of Properties table cannot be re-sized. It should be possible to widen the Description and use the scrollbar to read it.

[image]

nvdl, 2015-07-13:
Resizing the window does the job.

[image]

in3otd, 2015-07-14:
yes, I changed the table appearance and behavior to bring it closer to how it was in 0.0.17 and older in #232. IIRC, currently the table resizes the columns to fit all the text, except for the last column, which will be stretched or cut depending on the available space.
IIUC, now a horizontal scrollbar appears only if the window is made small enough so that there is not enough space to have the last columns starting in the visible area. Before, you had larger columns and often the last one (or two) were completely hidden and you had to scroll to realize there actually were more columns.
I guess this is another area where the expectations of new and experienced users differ: I am used to the small dialog size that Qucs had in the past - which allows to still see a good part of the schematic in the background - and usually I do not need to read the properties description for the most common components anymore, so for me it's important to see the complete content of the first two columns only. If I need to read the complete  description of a property I just enlarge the window, as nvdl pointed out - and here this does not happen often enough to be annoying. I understand a new user could want to read the properties descriptions more often.
A good compromise might be to have Qucs remembering the dialogs window size, as proposed some time ago by @nvdl IIRC.

guitorri, 2015-07-14:
On the above screenshot, for `tbhrec` one needs almost a full screen to be able to read its description.

If one needs to maximize to read it, we should add a the maximize button to the dialog/window.
 
At least the property editor on the right should have the Description in full. It should wrap the line around and not hide the text behind the `...`. 

We should make no assumptions about font and monitor sizes.

in3otd, 2015-07-14:
right, there are a few long properties descriptions around.
The only way to properly show them in the table will be to wraparound the text to the available width. But this should not be done automatically for every line, IMHO, as otherwise only a few properties will fit in the window height and you will need more scrolling up and down to access them.
Showing the description in full on the right side should be enough. And maybe add the complete description as a Tooltip when you hover with the mouse over the Description field in the table (no idea if and how it can be done, though)

nvdl, 2015-07-14:
Interestingly enough, If I select the property, the text is also clipped for the entry box.
I was expecting that to be the current fix.

in3otd, 2015-07-14:
? you mean actually the very first point in the initial message from guitorri ?

The clipping in the table and in the description over the entry line on the right are independent; the first is automatically handled by the Table widget, the second is "manually" done [here] (https://github.com/Qucs/qucs/blob/master/qucs/qucs/components/componentdialog.cpp#L634).

nvdl, 2015-07-14:
Yes, I forgot to read (remember) that point.
Why clip on the right side? There is enough space.
I think that can be the quick fix now.
Either the text box remains at one poistion or follows the height of the text.

Tooltip text is also a good idea.