Qucs/qucs #317: simulator handling from the user perspective

felix-salfelder, 2015-07-19:
support for different simulators mandates some corresponding user interface. it is pretty clear, that a simulator backend should be internally represented as a class derived from `Simulator` as introduced in #315. somewhere, we need a storage concept for the `Simulator*` pointing to the simulators presented to the user.

some questions arise (omitting possible answers intentionally).
- should the simulator set be global (per session?) or per schematic document?
- where should the simulator setup be stored? globally? simulator specific? per schematic file?
- should a simulator be represented by a schematic Element (if not, then how to embed simulator options into a schematic file?)

the implementation should be really straightforward, but please contribute more questions and some opinion...

ra3xdh, 2015-07-24:
Combined with #226 as a part of refactoring process.

felix-salfelder, 2015-07-24:
please don't hijack this issue. or make your plans clear.

imo there are only two options, how we can proceed, these are

* merge spice4qucs now and start refactoring from there
* don't merge spice4qucs now and block other works even longer.

i have already swallowed the first option, and i will no longer fight. what comes next?

ra3xdh, 2015-07-24:
I suggest don't merge `spice4qucs` now, refactor it , and then continue with merging of refactored `spice4qucs`.

felix-salfelder, 2015-07-24:
i will not work with unmerged or unreviewed code that i am not the author of. period.

in3otd, 2015-07-25:
Some comments to the original questions:
> should the simulator set be global (per session?) or per schematic document?

Per schematic. I would like to possibility to have a project with schematics using different simulators. Sure, there will be a default one set, when you create a new schematic but then it has to be tied to the schematic. If you open a schematic from the examples or received from someone else it makes no sense to have to select the simulator to use manually.

> where should the simulator setup be stored? globally? simulator specific? per schematic file?

You mean the simulator *environment* setup or the various simulation options? Environment setup (mainly simulator path) is a global setting (can maybe even support multiple versions of the same simulator). Simulator options like "use n threads" or "use binary output format" not sure, either schematic or global.

> should a simulator be represented by a schematic Element (if not, then how to embed simulator options into a schematic file?)

Unsure: different simulators have different options, even for "simple" stuff, like DC simulation. Maybe it will make sense to have different Simulation Components for the different simulators. I mean a "Simulator A DC simulation component", a "Simulator B DC simulation component", etc. Will make more difficult to change simulator though. But in this way you could also run different simulators at the same time on the same schematic (not sure who needs to to that routinely...)

felix-salfelder, 2015-07-27:
> Maybe it will make sense to have different Simulation Components for the different simulators.

i've been thinking about this. i'm convinced it will not. here is why: the schematic is supposed to formulate a problem, not how an attempt to solve it should look like. every simulator should have the same chance, and it's up to the simulator-glue to decide what to do about it. we should clarify what we mean by "do a dc simulation" and, starting from there, translate into commands for the various different simulators.

>  But in this way you could also run different simulators at the same time on the same schematic (not sure who needs to to that routinely...)

as i have posted somewhere else, i thought of extending the "enabled/disabled" flags for schematic elements to white or blacklist simulators (by name). this is still considerably hackish, but it's use would be optional. and its feels pragmatic: with these lists, you can add multiple "DC" simulations, each whitelisted for only one simulator.
(would that do the trick for you?)

ra3xdh, 2017-06-29:
This issues is related to #275, which is closed. Discussion thread at #316 is sufficient.

closed 2017-06-29 by ra3xdh