Qucs/qucs #318: Verilog-A modules builder concept

ra3xdh, 2015-07-19:
I would like to present my concept of Verilog-A modules builder. Test branch https://github.com/ra3xdh/qucs/tree/verilog-A contains source code. Verilog-A modules builder allows you to convert a Qucs subciruit (*.sch) to Verilog-A module (*.va). This concept was discussed with @MikeBrinson  about a month ago. At current state only EDDs and resistors are supported. There are no error-checks in schematic. Of course, there may be bugs.

How to use:
* Checkout and build branch
* Run Qucs, open subcircuit (*.sch)
* Click main menu *Simulation->Build Verilog-A module from subcircuit*
* Select filename and location to save module
* View module with test editor

Here is test circuit: https://gist.github.com/ra3xdh/0a579afdb7c24c003a79 . 

It is tunnel diode model.
[image]

And here is auto-generated Verilog-A code for it. 
~~~
`include "disciplines.vams"
`include "constants.vams"
module tunnsub(K, A);
inout K, A;
electrical _net0, K, A;
parameter real Is=1e-12;
parameter real Ip=1e-5;
parameter real Iv=1e-6;
parameter real Vp=0.1;
parameter real Vv=0.4;
parameter real K=5;
parameter real C=0.01p;
parameter real Temp0=300;
real VT;
analog begin 
@(initial model)
begin 
VT=(`P_K*Temp0)/`P_Q;
end
I(_net0,K) <+ Is*(exp(V(_net0,K)/VT)-1.0);
I(_net0,K) <+ ddt(C*V(_net0,K));
I(_net0,K) <+ Iv*exp(K*(V(_net0,K)-Vv));
I(_net0,K) <+ Ip*(V(_net0,K)/Vp)*exp((Vp-V(_net0,K))/Vp);
V(_net0,A) <+ I(_net0,A)*(1e-3);
end
endmodule
~~~

felix-salfelder, 2015-07-19:
this looks fun! it is consistent with #316. the code should go next to `verilog.cpp`, where the verilog netlister is implemented (or even share some code?).

> Click main menu Simulation->Build Verilog-A module from subcircuit

no. what we need is File->Export as.. and in the filetype dropdown menu, we should list the language extensions. #316 only provides a CLI option right now. the GUI thing should be implemented independently.

> V(_net0,A) <+ I(_net0,A)*(1e-3);

verilog-A supports subdevices. i'm inclined to propose that an export to verilog-A should faithfully reproduces subdevices, and not all subdevices can be easily expressed as a branch contribution. a user should choose an equation component instead of a subdevice if (s)he needs a branch contribution statement in the exported module.

guitorri, 2015-07-19:
@ra3xdh does this code has to be after spice4qucs?
I believe the `V() <+` contributions are not currently supported by the ADMS/qucsator interface...

ra3xdh, 2015-07-20:
@guitorri Yes, this branch is fork of `rebase_spice4qucs`. Development of Verilog-A modules builder and SPICE-support stuff will go in parallels.

> I believe the V() <+ contributions are not currently supported by the ADMS/qucsator interface...

No problems, I can use current-oriented form `I() <+` of equations instead. I saw such equations in `src/components/verilog` models used by Qucs via ADMS.

@felix-salfelder 

> no. what we need is File->Export as.. and in the filetype dropdown menu
Yes, I will move verilog-A exporter to File menu  group.

> verilog-A supports subdevices

I know about subdevices. But RCL can use not only constant values. There may be SPICE behavioral RCLs too. I think it's better to export them as equations.

I have plans to support at least the following components:
* EDDs
* RCL (usual and behavioral )
* SPICE B-type sources
* DC sources

These components are most used in compact models.

felix-salfelder, 2015-07-20:
> I know about subdevices. But RCL can use not only constant values. There may be SPICE behavioral RCLs too. I think it's better to export them as equations.

an `R` (same with `C` and `L` and some others) with a non-constant value is no longer a component of type `R`, which corresponds to a (constant) resistor in QUCS (and in verilog). it does not exist in QUCS, and it *must* not exist. unlike SPICE, QUCS uses different names for different devices. a QUCS device is much like a verilog device, fully specified by its type name. it is not a big deal to add a different component for different functionality. think of `VRect`, `VPulse` -- these work perfectly well, are not hacks, easy to understand, and still can be easily exported to some SPICE dialect *AND* to all other languages.

tl;dr:
* QUCS does not have to become SPICE to provide components that SPICE uses,
* QUCS must not become SPICE to maintain sanity.

we need standard compliance more than anything else, there's no point in moving further away for the sake of a particular and obsolete corner case (such as SPICE). don't pull in weird hacks here. these will break the future transition to verilog (or maybe the transition will break your code and the schematics for all the users).

thanks

ra3xdh, 2015-07-20:
> * QUCS does not have to become SPICE to provide components that SPICE uses,
* QUCS must not become SPICE to maintain sanity.

SPICE remains an industrial standard. And we should follow it despite our wishes. Many skilled engineers asked me about full SPICE compatibility in Qucs. Discussion at #275 will be continued later.

felix-salfelder, 2015-07-20:
i wasn't aware that i argued against spice *compatibility*. let me clarify.

the standard that is worth following is verilog, this is the one, the industry is using or heading towards to. QUCS should (for the sake of it) support both, verilog and spice, this is postulated by the U in the name. by poisoning the core with SPICE semantics (such as device "variants"), you thwart future development, and sane interoperability -- for nothing.

i have explained previously, how supporting SPICE does not interfere with the current component semantics that has been chosen (carefully!) by the original QUCS author. and i truly hope that you won't touch it.

thank you

PS: and please don't use the words "SPICE" and "standard" in a sentence without negation.

felix-salfelder, 2015-07-20:
as a side remark: what is wrong about letting the user decide how to create his model? that would mean (for the verilog-a netlister)
* user places component => user gets subcircuit component
* user places voltage+EQN => user gets `V(..)<+` statement
* user places current+EQN => user gets `I(..)<+` statement

in addition to that you could still add special SPICE components that behave completely different to that. in *addition*. and with a huge portability warning attached.

MikeBrinson, 2015-07-20:
The synthesis of Verilog-A code from Qucs EDD behavioural models is important because it fills in one of the missing links in the compact model development cycle.  The Qucs EDD structure has been designed to provide a one-to-one link between EDD and Verilog-A statements, see the Qucs presentations given at the MOS-AK workshops (http://www.mos-ak.org/).  Currently, new models can
be developed using EDD and iteratively tested via Qucs GUI.  However, an equivalent model described in Verilog-A has to be generated by hand !  The new synthesis engine will allow automatic generation of Verilog-A code from EDD, and other Qucs components and subcircuits.  As the ADMS Verilog-A to C++ compiler becomes further developed the generated C++ models will link directly to the Qucs, ngspice, Xyce and gnucap simulation engines.   For this route to fully work, much more work is needed on improving the subset of the Verilog-A HDL allowed by ADMS and also on improving the linking of the generated C++ models to all the above listed circuit simulators.

To assist with the generation of the Verilog-A code from Qucs EDD and other component models I
will post a series of modelling solutions to problems like how does one represent V(n) <+ I(n) and 
nonlinear inductors in Verilog-A.

MikeBrinson, 2015-07-21:
Hello Vadim
Thanks for taking time to push forward development of the Qucs EDD to Verilog-A synthesiser.After reading all the email on this subject I added a note on the GitHub column pointing out theimportance of this subject for compact modelling development.
Over the coming weeks I will contribute to the development of this topic, adding my knowledge ofthe Qucs EDD element, Verilog-A HDL statements and the current capabilities and limitations of the ADMScompiler software.  The recent release of Xyce 3.0 makes the development of the Qucs EDD to Verilog-A synthesisermost timely.  To make the synthesiser really effective we will probably need to make changes to the ADMS 2.34compiler and the Qucs XML scripts.  One obvious example is how we deal with linear passive components whengenerating Verilog-A code.  Other obvious Verilog-A statements to be dealt with include V(n) <+ 0.0 and V(n) <+ I(n)*R.
To start with I will try using your Verilog-A synthesiser code to generate Verlog-A models for a number of simple and more complexcompact device models. In addition  I will try develop a number of test cases which show the EDD schematics and the resulting generatedVerilog-A code for a number important modelling cases which are not currently implemented by ADMS.

Would you like me to send you directly my notes and test data or to simply enter them on the Verilog-A synthesiser GitHub web site?
Best wishes.
Mike

 Mike Brinson

mbrin72043@yahoo.co.uk

felix-salfelder, 2015-07-21:
> Other obvious Verilog-A statements to be dealt with include V(n) <+ 0.0 and V(n) <+ I(n)*R.

@MikeBrinson i am working on the left hand side potential contribution in gnucap-adms right now. maybe we should fix a common denominator in the implicit rules. how about the Qucs/ADMS issue tracker?

ra3xdh, 2015-07-21:
@MikeBrinson , the latest code at https://github.com/ra3xdh/qucs/tree/verilog-A branch contains current-oriented resistor model ( `I(...) <+ V(...)/R` ), capacitor model, and Menu point moved to *File* menu.

MikeBrinson, 2015-08-07:
The following diagram shows a set of experimental EDD/R/C  "templates" for constructing  more complex experimental compact Verilog-A device models.  These templates can be stored on a Qucs schematic (in a suitable project directory) and copied onto a working schematic when required.

[image]

These are just a small selection of possible template types which can be easily constructed.

As an example of their use consider the template for an inductance and the example of its use shown in the next diagram:

[image]

By combing the L EDD template with one external linear resistor an inductance model can be simply
formed: see next diagram
[image]

Overall testing of the new Verilog-A/R synthesised model is also straight forward - see next diagram.

[image]

MikeBrinson, 2015-08-07:
The unofficial Qucs releases Qucs-0.0.19S-rc1 and Qucs-0.0.19S-rc2 appear to have a bug in the naming of the circuit  ground node.  Some ground nodes were numbered 0 rather than gnd which in turn causes the ADMS compiler to fail.  I have pushed a block of C++ code to Qucs branch Verilog-A which should correct this problem.  The following test circuit and generated Verilog-A code suggest that this bug has been eliminated:
[image]

Please note the test circuit only generates a list of Verilog-A statements with changing node naming.
It has no other functionality.

[image]

MikeBrinson, 2015-08-09:
Added thermal current noise to resistors in Verilog-A synthesised modules.

The following example outlines how thermal current noise is handled by Qucs/spice4qucs when
synthesising Verilog-A modules from Qucs subcircuits. 
[image]

In this example a simple two resistor voltage divider is used to demonstrate the generation, and plotting, of thermal noise current and voltages.

Note how ground node terminals within a subcircuit are handled.  Verilog-A includes statement

ground gnd;

However, with the current Adms/qucs0.0.19 combination  Verilog-A statement "ground" is not fully
implemented.  As a work around  label the ground net, in a subcircuit,  gnd and connect it to a subcircuit interface pin.  This can then be connected to the main earth on a circuit schematic, ensuring that a ground net within a subcircuit is not LEFT FLOATING!  Also not with a high percentage of Verilog-A subcircuits a ground net is not defined.

ra3xdh, 2015-08-09:
@MikeBrinson  I have added improved fix for `gnd` bug by last commmit at `rebase_spice4qucs` branch. Now SPICE `0` node name should not appear in Verilog-A modules. Added Verilog-A/SPICE independent conversion for node voltages. Please test on your side.

Thank you for thermal noise and inductance models! I will integrate it during the next week.

MikeBrinson, 2015-08-11:
As an extension to the thermal noise model for resistors I have added models for semiconductor device shot and flicker noise.  The attached diagrams give details of the models and a simple test circuit.  Both the shot and flicker models require, as input, device DC bias current.  In response they output device shot and noise current in the AC domain.  In the test example a DC input current is
swept between 0 and 10A and the noise current measured and plotted as a function of frequency. 
Notice how the VA model takes Verilog-A statements as input to the noise model and outputs a working Verilog-A module.
[image]
[image]

MikeBrinson, 2015-08-13:
Added an EDD/Verilog-A CCVS to Qucs/spice4qucs.  This fundamental device is needed in compact device modelling to probe for aDC current in a branch and output a voltage with same magnitude as the probed current.  To test the operation of this device and check the operation of the Qucs/spice4qucs Verilog-A synthesiser a compact model of a basic semiconductor diode was constructed. The next three diagrams give details of the diode model and its operation based on the EDD schematic and the synthesised Verilog-A code model.  Note both forms of diode model give very similar results implying 
[image]
[image]
[image]

that the basic Verilog-A syntheiser and the new CCVS model are functioning correctly.

MikeBrinson, 2015-08-14:
Added CCCS, VCCS and VCVS linear controlled sources.  Test results for EDD and Verilog-A versions show below.
[image]
[image]
[image]

MikeBrinson, 2015-08-14:
More Verilog-A synthesised diode test results - temperature variation of diode properties and resulting I/V curves for EDD, Verilog-A and Qucs diode models:

[image]
[image]
[image]

ra3xdh, 2015-08-15:
Added inductance model and noise source model suggested by @MikeBrinson . Here is an example of usage:
[image]
Here is code example:
~~~
`include "disciplines.vams"
`include "constants.vams"
module Lsub(p1, p2);
inout p1, p2;
electrical p2, p1;
parameter real L1=1e-6;
parameter real Rleak=1e6;
analog begin 
I(gnd,_net0L1) <+ ddt(-V(_net0L1))+V(p2,p1); // Inductor: L1 
I(p2,p1) <+ V(_net0L1)/(L1+1e-15);
I(p1,p2) <+ flicker_noise(1e-6,0); // I1 noise source 
I(p1,p2) <+ V(p1,p2)/Rleak; // B1 source
end
endmodule
~~~

MikeBrinson, 2015-08-15:
More test results.  A long channel EKVv2p6 EDD based model.  This model demonstrates the use of EDD elements for simulating the non-linear DC I/V performance of an n type MOSFET.  The EDD model, synthesised Verilog-A code and typical test results are given in the following diagrams.  Please note that, at this stage in the development cycle,
[image]
[image]
[image]
[image]

 the generated Verilog-A code is not optimized for minimum run time.

MikeBrinson, 2015-08-17:
Attached is an optimised version of the EKV2p6 model:  NOTE (1) I(a,b) <+ 0; statements are now removed by the Verilog-A synthesiser, (2) the 1Ohm and 1e9 Ohm Verilog-A noise contribution statements have been edited out as these shouud NOT contribute to the device noise footprint, and (3) the EDD model has been minimised in size reducing significantly
[image]
[image]
[image]
[image]

 the number of contribution statements.

MikeBrinson, 2015-08-19:
Made improvements to ampere_noise.cpp and ampere_noise.h to allow shot and flicker noise sources to be synthesised from EDD compact device models.  The following diagrams show how to generate shot and flicker noise, their EDD templates and a set of synthesised Verilog-A code for the noise models.

[image]
[image]
[image]

MikeBrinson, 2015-08-22:
The Qucs inductance mode has been extended to include generation of Verilog-A code; see attached
results below for phase amplitude and phase results from a basic series LCR network.  An example of the Verilog-A code generated by the Qucs synthesizer for a multi-terminal LCR network is shown last.
[image]
[image]
[image]

MikeBrinson, 2015-08-24:
Made changes to the resistor, capacitor and inductance models to eliminate internal node references to ground node gnd.  he following test example shows a series of RCL components with different connections to earth.  NOTE this circuit is for test purposes and does not have a specific function.

[image]
[image]

MikeBrinson, 2015-08-25:
Changed ampere_noise model to remove node gnd references.  The following test circuit and generated Verilog-A listing demonstrates the removal of node gnd.
[image]
[image]

MikeBrinson, 2015-08-26:
Changed linear controlled source code to remove node ground (gnd) from synthesized Verilog-A modules.  The next two diagrams illustrate the process of ground node removal during Verilog-A code generation by the Qucs synthesizer.
[image]
[image]

ra3xdh, 2017-06-29:
Colsing. This Verilog-A module builder still be presented in Qucs-S http://ra3xdh.github.io/ . You can use it until the code will be backported in Qucs using modularity concepts.

closed 2017-06-29 by ra3xdh