Qucs/qucs #319: Component dialog move up/down fail to move row.

guitorri, 2015-07-20:
The move up/down on a Equation component only move items on the first column. It should move the complete row.

This messes up the left-hand-side and right-hand-side of the equations.

[image]
[image]

felix-salfelder, 2015-07-20:
is there any indication that this is related to the marker move cleanup? or am i just being paranoid?

guitorri, 2015-07-20:
:smile: Don't worry... this is related to `componentdialog`, the `slotButtUp` and `slotButtDown` to be specific.
Several dialogs had to be tweaked/ported to fit with Qt4, bugs were introduced in the process I think I was the last to mess with this dialog... I opened the ticked so we don't forget about it.

in3otd, 2015-07-20:
yep, as guitorri said, I've just checked, it was broken even before the marker move cleanup. (I did not have the time to bisect to find out the bad commit). So you can relax :smile:

guitorri, 2015-07-20:
I will take the blame on this again... 84f51e42

ra3xdh, 2015-07-21:
See #324 for fix

closed 2015-07-23 by guitorri