Qucs/qucs #321: Rendering differences: Export as image and Print (to PDF)

guitorri, 2015-07-20:
`Exporting to image` seems to render diagrams better than `Print` (and setting the output to a PDF). 
The rest of the schematic looks better on the `Print` output. 
The line thicknesses (dash spacing) and aliasing seem to be different depending on the print/export method.
I wonder if we can be more consistent.

See screenshots for the two outputs.

[image]
[image]

in3otd, 2015-07-20:
OTOH, I was wondering if it makes really sense to have an `Export as...` and `Print...` (in the main menu) which almost do the same thing.

The first picture seems to have anti-alias enabled, but using integer coordinates for the drawing, so the schematic comes out blurred (it reminds me of a similar effect I saw when working on the anti-alias stuff some months ago)

guitorri, 2015-07-20:
I think it makes sense to have both. `Print` is really meant for accessing the system printer settings. The `Export as image` does not actually print. Maybe we change the later to `Export as...` and then handle all the export types we might have in a custom dialog.

ra3xdh, 2015-07-20:
`Export as image` at first creates SVG and the passes it to `inkscape` in batch mode. `Print` uses `QPainter`. Inkscape is one of the best packages for publication-ready graphics preparation. And it provides the best quality of output.