Qucs/qucs #323: Qucs undefined symbol

jymag2, 2015-07-21:
When I try running qucs from the command line by using qucs [options], it says:
 qucs: symbol lookup error: qucs: undefined symbol: _ZN9QListData11detach_growEPii.
I installed qucs from tarball version 0.0.18. Using centos 6.6, qt-4.8.6. Is this how to open the GUI? Have you seen this problem before?
Thank you for your time and help.

in3otd, 2015-07-21:
I have tried building Qucs using the 0.0.18 tarball from Sourceforge in Ubuntu (Qt 4.8.0) and everything went fine; I could start the GUI from the command line by typing `qucs` and see the command line options with `qucs -h`.

Did you build previous/later versions of `qucs` before? I saw similar errors when building different versions and not cleaning up the installation directory before. I guess the build system somehow picks up old libraries instead of the new ones. It shouldn't happen normally, but I haven't been able to systematically reproduce the issue.

felix-salfelder, 2015-07-21:
`C++` symbols are scrambled, but you can

    $ c++filt _ZN9QListData11detach_growEPii
    QListData::detach_grow(int*, int)

`c++filt` is part of binutils.

`ldd $( which qucs ) |grep libQt`

tells you, which Qt libraries are linked to your qucs. you can use `readelf` to display the symbol tables of these libraries,

(yes, i don't know why your symbol is missing, but this might help you figure it out)

guitorri, 2015-09-28:
Please reopen if there are still problems.

closed 2015-09-28 by guitorri