Qucs/qucs #325: compilation time issues

felix-salfelder, 2015-07-22:
i have tried to compile and work with the spice4qucs branch. now this increases compilation time a lot. a huge part appear to be the components. but what's really bad is that everytime i change anything unrelated, all components get recompiled. i.e. at each step of a rebase.

we should not do these unnecessary recompiles. they can be best avoided by getting rid of the headers. we do not need one header per component, defining an interface that should actually be private, do we?

as a temporary solution i could add a switch to the build system to disable a subset of the components... any better ideas?

ra3xdh, 2015-07-24:
The title of this issue may be misleading. There are  no real compilation issues. 

Yes, numerous SPICE components were added. It's need to provide full SPICE compatibility and save Qucs legacy components.  Of course, it will increase compilation time. What else do you want? These components depend on `component.h`, `misc.h`, `node.h`, and `spicecompat.h` . I will look where I can remove unnecessary dependencies. What *anything unrelated* did you mean?

Combined with #226 and closed.

felix-salfelder, 2015-07-24:
> there are no real compilation issues.

who said that? on my end, theres a third word in the title.

felix-salfelder, 2015-07-24:
> What else do you want?

if everything else fails (and your header structure is optimal), i intend to add the switch disabling a subset of the components. exactly as i wrote (did i?)

ra3xdh, 2017-06-29:
Closing, because compilation issues were reported for #275. This PR is closed now.

closed 2017-06-29 by ra3xdh