Qucs/qucs #326: QEP: Multiple Simulators Support

nvdl, 2015-07-23:
[ ] - Base classes

nvdl, 2015-07-23:
@felix-salfelder 

    class Qucsator : public Simulator(){
      public:
        ~Qucsator();
        SimOutputData const* getResults(Schematic const*) const;
        SimOutputData const* getBiasPoints(Schematic const*) const;
        netLang const* netLang() const{return &qucsatorlang;}
        OutputData const* outputParser(QString filename);
      // and whatever else
      private:
       // internal stuff
    }

Why `getResults` is using schematic pointer?
I personally feel, every simulation run should have an ID (or a specific result file or a specific table in HDF5).
Of course, there should be a limit to the max number of simulation runs to keep.

Only `doSimulate` should get reference to the schematic.
Thus, `getResults` will only parse (or load) the simulation result based on the simulation ID.

If each simulation run can be identified uniquely, different runs can be compared.
Of course, it is doable with parametric sweep now but I am thinking out loud if there is something else to gain. Now, `qucsator` overwrites the previous simulation result.

`getBiasPoints` can be merged with `getResults` as it is related to D.C sweep. For other types of sweeps, should there be more functions? Or?

We should also add the ability to simulate using different simulators in one run to compare the results. Thus, the simulator type should be the part of the simulation declaration (e.g. transient simulation). This is coupled to the data representation as well (in terms of naming for the user) so that one can easily select the data set to plot. For example `TR1.Node1.Vt`, `TR2.Node2.Vt`.

felix-salfelder, 2015-07-23:
> Why getResults is using schematic pointer?

this is open for discussion. note that we do not only need a sane interface, but also something that works as a transition from the current state quickly. currently we need the schematic in oder to obtain results.

> I personally feel, every simulation run should have an ID

ok. but that is not part of the simulator class. something could easily do

     for id in 1,2,3
           results[i] = simulator->get_results(schematic);

you need to think of a place to put that (maybe attach a results deque to the `Document`)?

> (or a specific result filename or a specific table in HDF5).

definitely not. data must be provided independent of the retrieval implementation. think of a diagram, if you pass it a file name (as it is implemented right now), you need to reimplement it if you want to pass something else, and then you will have to reimplement it again and again.

rather, a diagram should get a pointer to a data object as in 

    void Diagram::attach(SimOutputData const *);

now, you can do

    x = mysimulator->getResults(...);
    mydiagram->attach(x["Ids"]);

or just

      filename = "whatever"; // obtained with a file chooser maybe
      x = dataFromFile(filename);
      diagram->attach(x["Ids"]);

the point is, that the diagram code works in either case. now there will be other sinks for data. think of postprocessing of some kind. you don't want to reimplement this every time, the file format or simulator changes, but just add another parser.

> Only doSimulate should get reference to the schematic and return the ID or the filename.
filenames are bad (as i explained above, hopefolly), but the exact semantics of how to organize the interface are open. i don't see the need for seperate `doSimulation` and `getResults` right now, but there might be one.

> getBiasPoints can be merged with getResults as it is related to D.C sweep. For other types of sweeps, should there be more functions? Or?

same here (it's completely open). i think the simulator is best to know what "bias points" even are. if i pass a random schematic to a simulator, it will be hard to figure out which parts of the data set even *are* bias points. so i added this preselection thing... it will get worse with multple dc simulations in the schematic, we are not there yet. think of a simple series of simulations first (e.g. one taking the circuit in the initial state for the next one).

> We should also add the ability to simulate using different simulators in one run

once we are able to simulate with a particular simulator *at all*, we are almost done. stash several simulators into a list attached to the Doc and run them all, even in parallel.

> Thus, the simulator type should be the part of the simulation declaration (e.g. transient simulation).
no. the simulation declaration (you are referring to the "Component" in the schematic?) must be independent of a simulator. selecting data will be done using dictionaries: i ask a simulator to simulate a schematic with a transient "tr1" and a dc sweep "dc1" and i get back a dictionary (the data set) that contains two data sets keyed "tr1" and "dc1" respectively.

> Simulator should be the property of the simulation block and not the schematic.
no, this is upside down. if a simulator cannot do the job, it must tell us. it is important to seperate the problem statement (netlist+commands) from the algorithm (the simulator). verilog does a good job in abstracting netlists, we need to add the commands in an unambiguous way. the current state is fair, but we should improve on it.

thanks for your input, lets hope this leads anywhere `:)`

guitorri, 2015-07-23:
I already mentioned that these are great ideas. This is an important refactoring and just have two remarks at the moment:

 - Like it or not, qucsator, iverilog and freehdl (perhaps ghdl) features must be preserved and prioritized.
 - The refactoring has to be signed-off with an automated regression test containing a good coverage of qucs projects.

felix-salfelder, 2015-07-23:
i am constantly trying to isolate obsolete code without changing its functionality. but we all know that this is a house of cards, and there will always be trouble...

i will clearly not be able to refactor all backends and glue code. before the qucsator interface is in place half way, there's no point trying to touch iverilog and freehdl. i *will* ask for help...

and i start right now:
if somebody feels like it, please implement the netlisters for iverilog and freehdl following the verilog example in #316, this will help to free the `Schematic` class, maybe others. repeat the same for xyce and spice (sorry for being sarcastic: what did you say about refactor upon the third time a feature has been reimplemented?)

PS: shouldn't we add some regression tests to the code base?

guitorri, 2015-07-23:
We should not be merging experimental code to master. We need to find another branching model. We are trying to keep master always deployable. The review will add delay and frustration. The current model is clearly not working and it is not scalable. Any ideas?

Shall we open a WIP branch for this? In this case it makes sense to give you push access. When do you need input you may send an email or make pull request for the WIP branch. 

Stale branches on the main repo are an issue, but that is for another rant.

Yes, we should add regression tests to qucs. The easiest is to start with the ".sch->.sch"  (load and save) and ".sch->qucsator netlist" functional checks. 
I think there is some initial support for this in qucs-test qucsgui.py module. It is not enabled in the CI build because I noticed some differences between running from the command line and doing the load/save interactively. Later we can add unit tests as the new framework evolves. 
Frankly my motivation with qucs-test was to put brakes on the sprint of development that was going on in qucsator in the past. Now we have to do the same for qucs.

Some are good at ignoring sarcasms others are aren't. In the end of the day we have to work together...

Eventually things will move into place.

felix-salfelder, 2015-07-23:
tests: #328

felix-salfelder, 2015-07-23:
> We need to find another branching model.
see #330?

nvdl, 2015-07-23:
>definitely not. data must be provided independent of the retrieval implementation

By that, I meant to return the identifier that can be passed to `getResults`.
Of course, it should not return data.

`getResults` needs to know which data set (file) to process if there are many results.
How to represent these results at the data file level is something that can be taken later; either a single data file or multiple data files. "Taken later" because, as a starting point, we should not modify simulators to spit out our data file. This is mainly coupled to `qucsator` that spits out text. Others are good at spitting out binary data files.

>rather, a diagram should get a pointer to a data object.

I agree, it is aligned with the previous discussions; internal data representation is unique.

>filenames are bad (as i explained above, hopefully

I was not clear; use the filename "text" as identifier (or hash etc.).
It is coupled to the above point; `getResults` needs to know the data set.

>it will get worse with multple dc simulations in the schematic, we are not there yet

Each simulation declaration can be directed to a separate simulation result (identified by its ID or hash). They can be run in parallel if independent.

>think of a simple series of simulations first (e.g. one taking the circuit in the initial state for the next one).

By this I assume, TR1 causes the circuit to end in a state that is the starting point of TR2.
I did not know it can be done. Do you have any example? Does `qucsator` support it?
All I can see is the initial voltage or current parameters for a component.

>no. the simulation declaration (you are referring to the "Component" in the schematic?) must be independent of a simulator

Yes, TR1, TR2, etc. I was looking at one object of `class simX : public Simulator()` per simulation component but you are thinking of one object per schematic.

Using the first approach, a schematic can have different simulation components being run by different simulators. In this way, all of the simulation components can also be configured to use one simulator. Using the second approach, that is not the possibility.
I do not know if that will ever be needed but it is good to think about these things thoroughly now.

All in all, it is about getting started with the code to get a feel of it.
I think you have already done some work on it. Do you have a baremetal branch now from where things can be taken forward? How do we merge to make sure that things are compatible and test?

I agree, we need an `unstable` branch that is not deployable but still can be contributed to.
Is there a risk of divergence from `master`? I do not know.

felix-salfelder, 2015-07-23:
> >  definitely not. data must be provided independent of the retrieval implementation

> By that, I meant to return the identifier that can be passed to getResults.
> Of course, it should not return data.

why would you ask the simulator for results if you dont want data back? what am I missing?

> getResults needs to know which data set (file) to process if there are many results.

yes. think of `Qucsator::getResults`. this is part of the simulator specific code. there you can do what you will, e.g. mess with file names. there is just no point in leaking the file name (none that i can think of).

> either a single data file or multiple data files.

no files. not one, not many. this is not required, and suboptimal.

> It is coupled to the above point; getResults needs to know the data set.

`getResults()` returns a pointer to the data set, so yes is needs to know about it.

> Each simulation declaration can be directed to a separate simulation result (identified by its ID or hash). They can be run in parallel if independent.

don't know what this is about. what is an ID for? what would a hash be for?

> Yes, TR1, TR2, etc. I was looking at one object of class simX : public Simulator() per simulation component but you are thinking of one object per schematic.

yes, the schematic must load without a simulator even attached. so the schematic components must be simulator independent, rather they express a "task". the simulator is just used to perform the task.

> Using the second approach, that is not the possibility.
> I do not know if that will ever be needed but it is good to think about these things thoroughly now.

okay, you propose to add one command to the deck that only one simulator is meant to process? currently, schematic components can be deactivated. what you want to do is deactivate for a subset of the simulators. this is really easy, once we *have* simulators and a set of simulators in the document context. the components and commands remain simulator independent.

> I think you have already done some work on it.

yes, it's not all ready yet. there is one important PR about markers with a strange problem #314. this should be independent of spice4qucs for the most part. i need an untesting branch to file more PRs, such as the rebased simulator branch #315.

thank you

nvdl, 2015-07-23:
>why would you ask the simulator for results if you dont want data back? what am I missing?

You have added `doSimulate` to the class and then `getResults`.
I am talking in context of your approach. If `doSimulate` returns the data, fine.
I do not see the reason to have `getResults`.

Otherwise `doSimulate` has to return "something"; be it a reference, ID or whatever for the `getResults` to parse. If there is always a single data file to parse, fine, there is no need to get anything from `doSimulate`; `getResults` will know which file to load from.

>don't know what this is about. what is an ID for? what would a hash be for?

As I said before, it is for accessing different simulation runs.
If it is not needed, it can be dropped.
I was thinking of having one run, change some components' values and run again and then compare the results. You can call it "semi-automated" parametric sweep.

felix-salfelder, 2015-07-23:
indeed. doSimulate is still there in the baseclass, i had it removed from the Qucsator example. my bad. in particular, my replies were not compatible with the wiki. sorry, my bad.

the reason i had added the doSimulate thing was to get the Schematic into play. don't know exacly. i think time will tell `:)`. i expect even more trouble in this area, like progress bars and dialogs. maybe i should make this fly at all. first.

> I was thinking of having one run, change some components' values and run again and then compare the results. You can call it "semi-automated" parametric sweep.

different simulation runs and keeping data in between is a valid goal. it does not belong into the simulator class. we could pile up results and by default have all widgets use the most recent. and we could add an option to keep at most k>=1 result sets. i think, we can safely postpone this.

nvdl, 2015-07-23:
>i expect even more trouble in this area, like progress bars and dialogs

That can possibly be handled by signals and slots if you meant outside interaction.

>the reason i had added the doSimulate thing was to get the Schematic into play

It is good to keep it as it can wait for the simulation to complete and emit signals while simulator is busy. The slots can update the progress bar or do other stuff. Possibly, this is the current implementation.

Regarding `getResults`, how to handle a situation where the data set is too big to fit in the memory?
If I understood correctly, it returns a reference to a data set object that eventually resides in the memory.

If there are 10 nodes and the user is plotting only one node, should all the nodes be loaded?

Do we need `getResult` or `loadResult` with a node (or probe) identifier?
Do we need `unloadResult` with a node (or probe) identifier to free up the memory?

felix-salfelder, 2015-07-24:
> how to handle a situation where the data set is too big to fit in the memory?

this is a good point. these things must happen behind the scenes, in the data set wrapper implementation.
a data set must only *provide* data. that does not mean that everything is in memory at the same time. think of (using hdf, .dat lacks the required index table)

     OutputData* F = myParser("file.hdf"); // does essentially nothing, just records a file name
     OutputData* D->getdataSet["TR1"]; // checks if TR1 is indexed
     assert(D); // this better exists.
     for( auto i in *D) {// actually access and read in data (through begin())
          ...
     }

again, nothing of this needs to be outside of the OutputData class, and there's no need to even know about this, in order to use the data.

> Do we need getResult or loadResult with a node (or probe) identifier?

no, the DataSet provides these, again

    OutputData* T OutputData* D->getdataSet["TR1"];
    assert(D); // this better exists.
    D->get("V(1)"); // or whatever else.

> Do we need unloadResult with a node (or probe) identifier to free up the memory?

we need to keep track of data set objects, if we want that or not. a functional front end could be used like this

     DataSet const* x = getsomething();
     Diagram* diagram = new Diagram(whatever);
     diagram->attach(x); // keep track of it
     // do something
     delete[diagram]; // also detaches data

     Diagram::~Diagram(){
           DataSet::detach(&data); // this is implemented in the DataSet base class
                                                   // (deletes the data if it's no longer used)
     }

hope this clarifies something...

nvdl, 2015-07-24:
Looks like we agree with the base structure. What is status of your branch? Are you waiting for someone to finish up (as I read through the other comments) or is there a branch where these things can be implemented right away?

As I feel, this approach will not break anything that has been done so far for multi-simulator functionality. Mostly, the code needs reorganization.

felix-salfelder, 2015-07-25:
we need #315... and there's some uncertainty about the integration. in my understanding, another venerable PR that is currently incomplete (and still growing) is considered more important. a quick rebase showed that there are plenty of merge conflicts. imo it does not make sense to move on.

on the technical side, there is a crash reported by @ra3xdh in #316. it must be related to the dispatchers but i could not reproduce it (nmi). in any case, we need to eliminate this crash.

thanks

felix-salfelder, 2024-07-03:
Baking simulators into the GUI is asking for trouble. We need to support a standard for data interchange between tools.

closed 2024-07-03 by felix-salfelder