Qucs/qucs #327: Tuner [Feature request]

arhiv6, 2015-07-23:
[Feature request]
Hi all. AWR is a handy tool - tuner. It allows you to interactively change the parameters of the components and watch the changing characteristics: http://www.youtube.com/watch?v=oNnjEAoYqkQ
How can you implement a simple tuner in Qucs? Qucs has a simulation mode "Parameter sweep". Now the simulation in this mode, the graph shows all data sets, and if a lot of them - nothing is clear. Can add to the panel next to the "Set marker on graph" button "Tuner", a call which opens a small window that is similar to:
[image]
If scheme has  "Parameter sweep" blocks, after the simulation calculated data are in .dat file. When you change the slider new data set is shown in the chart (without delay on the calculation).
Pseudocode: (/qucs/dialogs/tunedialog.cpp)
```
TuneDialog::TuneDialog(Schematic *Doc_)
			: QDialog(Doc_)
{
  Doc = Doc_;
  setWindowTitle(tr("Tuner"));

  QGridLayout *all = new QGridLayout(this);
  all->setMargin(5);
  all->setSpacing(5);

  QLabel *lab1 = new QLabel(tr("Parametr"));
  all->addWidget(lab1, 0,0);

  QLabel *lab2 = new QLabel(tr("nom"));
  all->addWidget(lab2, 1,0);

  QLabel *lab3 = new QLabel(tr("max"));
  all->addWidget(lab3, 2,0);

  QLabel *lab4 = new QLabel(tr("min"));
  all->addWidget(lab4, 4,0);

  for(Diagram *pd = (Doc->Diagrams->first()); pd != 0; (pd = Doc->Diagrams->next()))
  {
    foreach(Graph *pg, pd->Graphs)
    {
        for (int i=1; i< pg->cPointsX.count();i++)
        {
            qDebug() << pg->cPointsX.at(i)->Var << pg->cPointsX.at(i)->count;
            double min=pg->cPointsX.at(i)->Points[0] ;
            double max=pg->cPointsX.at(i)->Points[pg->cPointsX.at(i)->count-1];
            QLabel *lab5 = new QLabel(QString::number(max));
            QLabel *lab6 = new QLabel(QString::number(min));
            
	    if (max>min)
            {
                QLabel *lab7 = new QLabel(QString::number(min));
                all->addWidget(lab7, 1,i);
                all->addWidget(lab5, 2,i);
                all->addWidget(lab6, 4,i);
            }
            else
            {
                QLabel *lab7 = new QLabel(QString::number(max));
                all->addWidget(lab7, 1,i);
                all->addWidget(lab6, 2,i);
                all->addWidget(lab5, 4,i);
            }

            QLabel *lab = new QLabel(pg->cPointsX.at(i)->Var);
            all->addWidget(lab, 0,i);

            QSlider *sl = new QSlider(0,pg->cPointsX.at(i)->count,1,0,Qt::Vertical);
            all->addWidget(sl, 3,i);

            connect(sl, SIGNAL(valueChanged(int)), SLOT(slotNewValue(int)));
        }
      }
  }
  show();
  return;
}

void TuneDialog::slotNewValue(int)
{
	QFileInfo Info(Doc->DocName);
	QString DataSet = Info.dirPath() + QDir::separator() + Doc->DataSet;

	for(Diagram *pd = (Doc->Diagrams->first()); pd != 0; (pd = Doc->Diagrams->next()))
	{
		foreach(Graph *pg, pd->Graphs) 
		{
			// update Graph realization
			pd->updateGraphData();
		}
	}
  	Doc->viewport()->update();
}
```
Anybody can implement such a feature?  (Sorry, my english isn't good).

guitorri, 2015-07-23:
There is a user in SourceForge with something along this lines:
http://sourceforge.net/u/tipofthesowrd/qucs/ci/tunerdialog/tree/
He communicated a few times on the qucs-devel list. I am not sure about the status.

in3otd, 2015-07-23:
...last time I checked that Tuner dialog implementation, it was*mostly* functional, even if somewhat prone to crashing in some situations. My feeling was that there was not much work needed to finalize it but did not have time to dig into that.
Note that it restarts a new simulation every time a parameter is changed, it does not (re)use the simulation sweeps.

arhiv6, 2015-07-23:
I suggest not to run simulation after changing settings in the tuner. I propose to start the tuner after the simulation, if the scheme has a sweep parameters. The data already in the file .dat, when changing the tuner loaded on the chart the necessary data.

in3otd, 2015-07-23:
resimulate after each change seemed ok to me, maybe also because at least one commercial tool does this. Of course it has the disadvantage of increased "latency" to see the new results when you change a parameter, so it works well when you have a small circuit and/or a fast machine.
Your solution also makes sense, if you know upfront which are the parameters limits and steps to use, otherwise you might need to change the sweeps limits and steps several times before finding the right parameter range to explore.

So I think both options could be made available: I will call the first one (re-running simulations at each step) "tuning" and the other (display only a part of the results of [multiple] sweeps) a "data display filter".

tipofthesowrd, 2016-02-14:
I tend to agree with @in3otd regarding "tuning" and "data display filter".
Tuning for me, at least as it is done in 2 commercial packages is a step change of a parameter and where several parameters can interact. The user needs to be able to at his convenience step through the different values. Sometimes up in value, sometimes down.

If you simulate all possible combinations before making the data available this becomes really intense of you have several parameters you are changing.