Qucs/qucs #328: regression tests for qucs

felix-salfelder, 2015-07-23:
imo we should add tests right next to the code. tests should be arranged in a test suite, and invocation should be easy. the GNU standard way is a check target.

    $ ./configure
    $ make check && echo everythings fine

this is implemented  in automake, terribly simple and pretty maintainable. i guess something similar is also available for cmake, althogh hand coding would not involve rocket science.

we should start tests with netlist reproduction and conversion, later we should also add pngdiffs and pdfdiffs against optical output, if this is feasible. test should be quick and simple...

(maybe there should also be numerical regression tests added to qucsator, but this is a different issue)

any thoughts?

guitorri, 2015-07-23:
Tests right next to the code are just fine.

guitorri, 2015-07-23:
I deliberately put qucs-test aside to ease bisection. Clearly it does not apply here because there is no way to operate older qucs from the command line. It is already as a submodule (in the wrong place imo). The intention was to (short) available examples also for test (and later for user documentation and what not).

felix-salfelder, 2025-03-02:
We have added a "tests" directory with a bunch of tests (predominantly from qucs-test) because it was needed. It remains to clarify the relation to "examples", but there is no urgency.

closed 2025-03-02 by felix-salfelder